mike4192/spotMicro #1: Servo Parameters

sffranke, 2020-07-02:
Thank you for sharing your outstanding project!
Can you tell me how you get to the parameters and how the direction of rotation of the servos is taken into account. I am experimenting with MG92b servos, which unfortunately do not all have the same direction of rotation.

RF_3: {num: 1, center: 306,range: 385, direction: 1, center_angle_deg: 84.0}

I hope I have understood it correctly. 
Here the motor turns from the center position (= 90 degrees) by another 84 degrees in positive direction of rotation (counterclockwise), which is the idle position. What does the range say in this context? 

Thanks in advance
 Steffen

mike4192, 2020-07-02:
Hi. Excellent question, I admit I did not provide adequate information to build up the servo configuration dictionary. I'm currently on vacation till Sunday, but I'll try and provide some textual information for now. When I get back I plan to update the readme with instructions and diagrams on how to build up the servo dictionary.

sffranke, 2020-07-02:
Great, looking forward to it. Enjoy your vacation!

mike4192, 2020-07-02:
I created a spreadsheet to help create the configuration values, and I did a servo calibration by manually moving the servos of each joint (with the servo_move_keyboard node) to reference positions (that I mostly measured by eye, for example horizontal and vertical), recording the raw absolute command values.

I'll first try and define the values in the servo dictionary:
Center: the "raw" pwm value the pca9685 node recieves, representing the servo center position. For background, a servos position is generally commanded by a 1 to 2 ms pulse in a 50 hz cycle (20ms time period). The PCA9685 board and ros node controls the length of this pulse via a 12 bit pwm value. Assuming a 20ms total cycle length, 0 means no pulse, 2^12=4096 means constant high signal, and 2048 for example would mean at pulse length of 10ms. A servo's center position is usually positioned with a 1.5 ms pulse which cooresponds to a pwm value of about 307. This value can be fine tuned by moving a sample servo with a horn on it via the servo keyboard move node and findibg the center point from where the servo moves equally in opposite directions (say +/- 90 deg equally from the center. But 307 is probably fine, and should be the same for all the servos in a set assuming they are the same kind.

mike4192, 2020-07-02:
Range: the "raw" pwm value that cooresponds to the max end to end movement range for a servo. This will be related to the max servo angle defined in the config file. Usually robotic servos can move +/- 90 deg, but I set a somewhat constrained value, I think 80 deg, to protect the servo from being commanded too close to it's command range. So for example, the servo range value would be the pwm value for a 80 deg servo position minus the pwm value for a -80deg servo position. The spreadsheet I will eventually upload automatically computes this value after calibration.

mike4192, 2020-07-02:
Direction: A value of 1 or -1. Reverses the direction or servo rotation if needed, dependent on the coordinate axes of a specific leg joint.  It is likely any others servo installee using the same spot micro frame and same servo orientation will have the same servo direction values as my dictionary.

Center-angle-deg: value in degrees, in a leg joint's specific coordinate system, that the servo is positioned at at it's "center" pwm value. Value outputted by a spreadsheet I will eventually upload

FreeBSDGeek, 2020-07-02:
Wow,  I was just about to ask the same question.  Thanks for getting back so quickly Mike.  I look forward to hearing more on how to setup the servos.  

I think I have a handle on the centre and and range values but I might need a diagram or something to help explain where the resting points are expected to be for each servo.  I installed and celebrated my servos originally as shown in the SpotMicroAI docs here: https://gitlab.com/custom_robots/spotmicroai/basic-instructions/-/tree/master/4.%20Auto%20boot%2C%20configuration%20file%20and%20calibration

Dan10110, 2020-07-06:
Hi Mike,

Thanks for sharing your code. It produces very good gait as shown by your video. I have followed your instructions and got your code to compile and run. I configured the servos using your servo tool as best I can. I got my Spotmicro to come alive yesterday!! Thank you again. It moves but not as nice as yours. I have a couple of questions.

1. Center angles- The knee and hip angles are easy to figure out since there are no other reasonable choices. However, the shoulder angle could be referenced to horizontal or vertical. Which one is it? Please see attached picture.

[image]

2. In your video, you demonstrated 2 gaits. The 1st one move 2 legs at a time and the 2nd one is 8 phases of one leg at a time. The yaml file is set to 8 phases. I couldn't find an option for the 1st gait style which seems to have a more stable motion. Is there an option for it?

Sincerely,
Dan

mike4192, 2020-07-06:
Hi Dan.

Congrats on getting my code running on your spot!!! I am amazed and humbled. I am working on a comprehensive guide with diagrams and sample step by step instructions for the servo/ leg link calibrations for the servo config dictionary. I hope to get some of that info uploaded later tonight.

Regarding your question for the alternative gait, that currently exists in the branch named alternate_gait (not to be confused with obsolete alternate_trot branch, will try to delete that one).

Dan10110, 2020-07-07:
Hi Mark,

I got the 4-phase gait running. I now understand why it is in the alternate branch. The hip doesn't move and that contribute to the apparent  motion stability. But why have a servo if it doesn't move. Besides, the 8-phase gait looks cooler. Thanks for pointing me to the directory. Should have figured it out.

I had encounter 3 or 4 minor issues in getting your code to run. I will open up another Issue to post them with the solutions so others can have an easier time.

Thanks,
Dan

mike4192, 2020-07-07:
Hi all. I am pleased to report that I have update the repo with comprehensive servo calibration instructions. If no more questions arise, I shall close this issue.

@Dan10110, do please provide feedback of other issues you had getting my code to run. I will try to address them.

FreeBSDGeek, 2020-07-07:
Thanks Mike,  The servo calibration instructions are great.   I read them over, very clear and I'm just going though the calibration with my Spotmicro.  I really appreciate the time you took in putting this together.

Dan10110, 2020-07-08:
Hi Mike- I didn't expect your calib procedure to be this detailed. It is very clear and easy to follow. It makes a lot of sense. I re-calib my Spotmicro following your procedure and now I am a happy man. My Spotmicro walks straight and moves like yours. I tried all 6 key controls and what a wonderful sight. There will be many more happy men. Thank you for the great work.

closed 2020-07-08 by mike4192

sffranke, 2020-07-09:
Mine woks fine too, Thank you Mike!