mike4192/spotMicro #3: Add custom 3d part models

mike4192, 2020-07-08:
I modified some of KDY's parts to improve joint stability for the shoulder joint (in the original model, it is supported only on one side of it's axis). This issue is to add my custom parts (or links to them on thingverse) and some description of how to print them, and additional hardware required.

Hardy5012, 2020-08-19:
Does cls6336hv files have solved this issue (in the original model, it is supported only on one side of it's axis)

rafaturtle, 2020-08-28:
Wondering if you can also share the plate with the screw placements for the Rpi. thanks heaps!

mike4192, 2020-09-14:
Just merged in a updated readme that has thingverse links to my custom shoulder assembly parts, as well as RPi and servo board platforms I created.

closed 2020-09-14 by mike4192