mike4192/spotMicro #4: Wiring

binaryhellstorm, 2020-07-29:
Do you have a wiring diagram for how you hooked up the Pi and servo controllers?

OttPeterR, 2020-07-29:
Take a look around [here](https://gitlab.com/custom_robots/spotmicroai/electronics) for some ideas.
There's no single way to do it, so you can see what you'd like to do and wire it up yourself!

sffranke, 2020-07-29:
I use 2 x PCA9685 too due the current the servos draw.

mike4192, 2020-07-29:
Hi, I do not have a wiring diagram, if I have time I'll try to create a basic one.

The wiring is fairly straightforward, I use one PCA9685 board, but I've heard people use two. This code would need to be modified to work with two boards. I have also heard of the potential current restriction issues, so I shorted the input servo voltage lines directly to the servo power rails to circumvent any such fuses. I've not experienced any issues with this setup, but I guess bypassing a fuse does pose the risk of burning out a servo from over current.

On my spot I have simple power distribution board with headers for ground and +5 V , sourced from a 5 V UBEC powered directly by a 2s lipo.

The Raspberry pi is connected to this 5v power via pin 2, and ground to pin 6.
PCA9684 SDL and SCL pins are connected to corresponding Raspberry pi SDA and SCL pins (pins 3 and 5, respectively).
PCA9685 VCC is connected to +5v power, and GND to ground. V+ port is left unconnected (this is the same as the servo power), as is the 0E pin.
PCA9685 Servo power input (green power block) connected direct to 2s lipo.

The servos are connected to ports 1 through 12 in the order described in the readme.

sffranke, 2020-07-30:
Thank you for your detailed reply! Mine works with two PCA9685 fine without any modification of the code.

binaryhellstorm, 2020-07-30:
> Hi, I do not have a wiring diagram, if I have time I'll try to create a basic one.
> 
> The wiring is fairly straightforward, I use one PCA9685 board, but I've heard people use two. This code would need to be modified to work with two boards. I have also heard of the potential current restriction issues, so I shorted the input servo voltage lines directly to the servo power rails to circumvent any such fuses. I've not experienced any issues with this setup, but I guess bypassing a fuse does pose the risk of burning out a servo from over current.
> 
> On my spot I have simple power distribution board with headers for ground and +5 V , sourced from a 5 V UBEC powered directly by a 2s lipo.
> 
> The Raspberry pi is connected to this 5v power via pin 2, and ground to pin 6.
> PCA9684 SDL and SCL pins are connected to corresponding Raspberry pi SDA and SCL pins (pins 3 and 5, respectively).
> PCA9685 VCC is connected to +5v power, and GND to ground. V+ port is left unconnected (this is the same as the servo power), as is the 0E pin.
> PCA9685 Servo power input (green power block) connected direct to 2s lipo.
> 
> The servos are connected to ports 1 through 12 in the order described in the readme.
> 
> I additionally have a master power switch for everything, as well as a secondary power switch that only cuts servo power (but keeps the raspberry pi and other electronics powered).

Thank you that's exactly what I was looking for. I missed that part of the readme.

closed 2020-09-09 by mike4192