mike4192/spotMicro #6: PS4 Controller

sffranke, 2020-08-27:
Hello again, t see that he code mentions a PS4 Controller, can you please give a hint how to use it?

mike4192, 2020-09-09:
Hi,

Unfortunately my project does not support a PS4 controller. The remnants of code you see referencing a PS4 controller is likely from code I used as inspiration from the stanford robotics club. Please see the link at the bottom of the main readme file for a link to their source code.

Unfortunately I can't provide much info on how to use a PS4 controller. I would suggest finding code that gets input from the PS4 controller and which would update the values in the command class.

closed 2020-09-09 by mike4192

sffranke, 2020-09-09:
Hi Mike,

thank you for pointing the right direction!