mike4192/spotMicro #7: A few things i am having trouble with.

Legoviathan, 2020-09-05:
so being 13, i don't get everything about this project. like, how to charge the battery. the battery comes with a charger, but to connect it directly to the servo control board means i have to snip off the connector thing. and then i cant charge it. do i have to buy a connector to mount on to the board? another one is, how do you control it via laptop or other device? i saw in the video that you were using a laptop to control Spot Micro, and i was wondering, was that just a bluetooth connection? or was it another linux device? Thanks a bunch!

mike4192, 2020-09-09:
Hi,

Thanks for the interest in my project!

I am using a lithium polymer battery (also called a lipo battery), which can be charged by a lipo battery charger. I use a charger similar to the B6 V2 Balance charger (just google it and you will find many similar ones, they are all basically the same).

The battery has a XT30 female connector which I use to both charge it and to plug it into my robot. Lipo batteries have a lot of different connectors these days, but the most popular ones currently are XT30/60/90 (the number refers to the max current it supports), deans connectors, and EC3 connectors. My wiring is a little complicated because I have some switches in line so I can power the robot off or on, but basically there are two wires going to the servo control board which I soldered a male XT30 connector to. The wires also go to a 5V UBEC, which steps down the battery voltage to a 5V output, which I use to power the electronics, such as a the raspberry pi. That's done through another small board I soldered with standard pin headers so I could use bread board wires.

Basically, a project like this will require a decent amount of soldering, some basic solderable bread boards, wires, and connectors.

The robot is controlled by my laptop keyboard. The laptop is running linux which is running a ROS python program which takes in keyboard button presses and converts them to speed or orientation commands. Both my laptop and the raspberry pi on the robot are connected my home router's wireless network. The commands are transmitted across the network, so no bluetooth is used. The data itself is transmitted via multicast UDP packets I believe.

Let me know if you have more questions!!!

Legoviathan, 2020-09-09:
ok thank you for your previous answers. i am awaiting parts that i can test on the 11th. so another question i had was can i use PuTTY to establish an ssh connection, and run multiple sessions? or can i use remmina and run the wireless screen? how exactly do i do that? and also, im wondering how i can merge a few nodes. because, is there a standing node? if so, in that node, can it rotate on its joints on place? and if it can, i want to run 2 different nodes, using a camera and face detection on one, and the standing node on the other and have it follow a face. is it possible to run those nodes separately, or do i have to merge the code? thanks a bunch!

Legoviathan, 2020-09-09:
oh and by the way, ive really wanted a unitree a1, so when i saw this, i remembered i had a raspberry pi. so i invested every penny i had into it. thanks for giving me something to work on!

Legoviathan, 2020-09-11:
and also, does your robot use a gyroscope or accelerometer?

mike4192, 2020-09-14:
You can use PuTTY to establish a ssh connection to the RPi. Make sure your RPi is connected to the same network, and then first start out pinging the rpi's ip address to make sure your computer can see it.

I haven't used putty that much, so I'm not sure if you can run multiple instances to have multiple ssh sessions. But as an alternative, you could install tmux onto the pi (a terminal multiplexer), and then from one ssh session you could effectively have multiple terminal windows. You'll need to look up the keyboard commands for using tmux, however, like for splitting panes and navigating windows. I would generally not recommend remina as it will probably be pretty slow, I would first connect the pi to a monitor and keyboard via a hdmi cable.

If you were to make a camera tracking node to send commands to the robot, you would not have to merge the code, the nodes could remain separate. Basically, a camera face tracking node could send commands to the spot_micro_motion_command node to stand the robot up, enter angle command mode, and then command the robot angle. The documentation I've written unfortunately isn't very detailed about the exact interface the commands go through, but if you look through the python node, spot_micro_keyboard_command, you should be able to see how the robot is commanded through its states.

In order to run ROS on the raspberry pi's own linux distro (raspbian), you would have to compile the ROS kinetic source code from source on the raspberri pi itself. Unfortunately that is a pretty complicated and tedious process and I really wouldn't recommend it. I tried doing it myself but quit because it was taking too long to figure out when an existing OS image was available with ROS prebuilt and ready to go.

If you look at the readme's "Control Instructions" section, there is at least some basic information of the steps of commands to get the robot to stand and transition through the state machine.

My spot does not currently have a gyroscope or accelerometer, but I'd like to incorporate one some time in the future.

mike4192, 2020-09-14:
BTW, I would recommend going through some of the ROS tutorials to get a handle on working with ROS nodes:
http://wiki.ros.org/ROS/Tutorials

Legoviathan, 2020-09-18:
ok this should be some of my final sets of questions. so when i type "rosrun i2cpwm_board i2cpwm_board" it spits out [rospack] Error: package '12cpwm_board' not found. and this command has to be run before all of the others right? so if i cant run that, i cant run anything. so pls explain that to me.

Legoviathan, 2020-09-18:
and i downloaded ubuntu on the pi and is working excellent otherwise

Legoviathan, 2020-09-19:
and one more, when i try to download the repository recursively, it wont let me. it will say "Permission denied (publickey). fatal: Could not read from remote repository" how should i bypass this?

Legoviathan, 2020-09-20:
ok i have figured out how to do all of the remote repository stuff, so now i am having trouble with the rospkg package. the computer says it cant find it, i know it is there. please help me

mike4192, 2020-09-21:
To address possible things to look into about your question about the i2cpwm_board not running, this sounds like your ROS workspace is not set up correctly. Do you have ROS installed on the raspberry pi? Have you sourced the setup.bash file to setup your catkin workspace? My catkin workspace is located in my home directory, so the way I set it up is by running the following at a terminal (this command will not output anything).
`source /home/mike/catkin_ws/devel/setup.bash`

This needs to be done for each terminal before you try running any local ROS packages. It is possible to automate this.

Note that the i2cpwm_board node will not run successfully if an actual PCA9685 board is not powered and connected to the raspberry pi (however it should still get past your error before then).

As for your second question, it sounds like you figured it out, but if not, you may have not setup a ssh key for your raspberry pi maybe? You can follow this guide to do so: https://docs.github.com/en/enterprise/2.15/user/articles/adding-a-new-ssh-key-to-your-github-account.

closed 2021-01-02 by mike4192