mike4192/spotMicro #9: Can't download the spot_micro_motion_cmd/libs directory

fjblau, 2020-09-12:
Not sure what is going on but it doesn't copy with the clone and when I try and sync dependencies I get a permissions error

kejincan0527, 2020-09-15:
When you download Mike's code for the first time, come to Mike‘s project folder
```
git clone --recurse-submodules https://github.com/mike4192/spotMicro.git
$cd spotMicro
```
modify .gitmodules
```
vim .gitmodules
```
chang url `ssh` to `https`
git@github.com:mike4192/xxx.git
to 
https://github.com/mike4192/xxx.git

and run
```
git submodule sync
git submodule init
git submodule update
```
You will get the complete project.

mike4192, 2020-09-15:
Thank you for commenting on elaborated sub module checkout instructions. I plan to add these instructions explicitly to the readme soon.

PR #12 (2020-09-21): Added additional install and run instructions
Closes #9 and #2.
closed 2020-09-21 by mike4192