mike4192/spotMicro #11: wire frame error

niranjan1997, 2020-09-15:
Even after installing all the subfolders of the repo whenever I run the node **rosrun spot_micro_plot spotMicroPlot.py**  I am getting this error 

`_ubuntu@niranjan:~$ rosrun spot_micro_plot spotMicroPlot.py
Traceback (most recent call last):
  File "/home/ubuntu/catkin_ws/src/spotMicro/spot_micro_plot/scripts/spotMicroPlot.py", line 29, in <module>
    ax.set_proj_type('ortho')
AttributeError: 'Axes3D' object has no attribute 'set_proj_type'
ubuntu@niranjan:~_`

please help me fix this error.
thank you.

mike4192, 2020-09-15:
This sounds like your running with an old version of matplotlib. Try updating matplotlib on your machine. I'll try and look into how to implement a version check.

Try the following command: `python -m pip install matplotlib --upgrade`, and/or follow prompts in your terminal, it may first ask you to update pip itself.

Note, usually best practice is to install and update python dependencies in a python virtual environment, however I'm unfortunately unsure how ROS and python virtual environments can mesh together. Since I've been working on a RPi and my personal laptop, I've felt comfortable updating matplotlib globally.

niranjan1997, 2020-09-16:
Thanks, Mike updating the Matplot lib to the latest version solved the issue

niranjan1997, 2020-09-16:
[image]
Hey Mike, when I ran this command `rosrun spot_micro_plot spotMicroPlot.py` 
I am getting this blank figure. what could be the reason. please help me fix it.
thank you.

niranjan1997, 2020-09-16:
This issue I solved, it was my mistake I haven't run this `roslaunch spot_micro_motion_cmd spot_micro_motion_cmd.launch` 
thank you.

closed 2020-09-16 by niranjan1997