mike4192/spotMicro #14: spotMicroPlot.py display empty

EastwoodStudio, 2020-10-07:
Firstly, I got then same trouble as #11 .
After run command _python -m pip install matplotlib --upgrade_ , still got error about numpy.
Then I upgrade numpy to 1.16.4, error was gone.
But when I started RPI nodes, then run spotMicroPlot.py on pc(virtualbox 6.1 ubuntu 16.04.7)
it looks like the next pic:
[image]
strange thing is the when I use Ctrl+C to end the program, matplotlib shows the dog status picture
[image]

I'm not familiar with matplotlib, please give some help, thx

info maybe useful:
matplotlib version 2.2.5
numpy  version  1.16.4
matplotlib backend: TkAgg

mike4192, 2020-10-08:
To confirm, are you seperately running the spot micro motion command node? (For example, via `roslaunch spot_micro_motion_cmd spot_micro_motion_cmd.launch`)

If you run this on a PC or virtual machine, it will probably error out as it will fail to complete the servo configuration call. This error can be bypassed by setting the `debug_mode` option from false to true in the yaml configuration file for the spot_micro_motion_cmd node.

EastwoodStudio, 2020-10-08:
Finally, I found out why.
spot_micro_plot node runs in virtualbox named as ros-pc. Executes _rosnode list_ on RPI I can see /spot_micro_plot. Then I tried to execute _rosnode info /spot_micro_plot_, it show me communication error with pos-pc.
That's because I set network of virtualbox as NAT mode.
After changed it to bridge mode, spot_micro_plot displayed very well :)

Thanks for your reply anyway @mike4192

closed 2020-10-08 by EastwoodStudio