mike4192/spotMicro #15: Servo Torque

fjblau, 2020-10-11:
I have successfully built this amazing robot, even the extended shoulder supports!

Raspberry Pi and ROS are all working fine... I went through the calibration and while it drifts around a bit, that seems to work ok. 

I am using the recommended PDI-HV5523MG servos with a 7.4V LiPo RC battery.

The problem is that the robot seems to be too heavy for the servos to lift when I issue the "stand" command. It sort of struggles a bit to get 3 legs up, but the 4th leg (RB3) never fully extends unless I put the robot on a stand without adding any weight. On the stand, the legs seem to extend to the proper position.

Is this a design or hardware issue? Or is there a better servo/power combination that I should be using?  Any advice is much appreciated! And I really enjoy this project!

EmperorKonstantin, 2020-10-12:
### Servo Motor Producing Reduced Torque

Torque is the amount of muscle, energy or power required to rotate and move a mechanism. It comes from three basic sources: friction, external forces like fighting gravity and accelerating a mechanism’s inertia.

By nature, motors will only put out a certain amount of torque so that if you choose the wrong one, it may not be able to handle the workload your team needs it to. Sometimes, you may also have a servo motor malfunction, and it will cease producing enough torque. Some of the most popular servos operate between 4.8v and 6.0v which is 130.5 oz-in and 152.8 oz-in respectively.

Aside from the motor not having enough power, here are some other things you can check:

- The servo amplifier rating.
- The servo’s workload limits.
- Low voltage or voltage drops within the system.
- Bad capacitors.
- Inadequate or improper connections.
- You and your team will want to become familiar with how to test servo motor and drive systems because this can often happen, especially with older and dated equipment.

Source: https://gesrepair.com/servo-motor-drive-troubleshooting-guide/

There are some pretty in depth resources for troubleshooting servos here:

- [Common Motion Problems and How to Fix Them - Part I](https://www.pmdcorp.com/resources/type/articles/get/common-motion-problems-and-how-to-fix-them-part-i-article)
- [Common Motion Problems and How to Fix Them - Part II](https://www.pmdcorp.com/resources/type/articles/get/common-motion-problems-and-how-to-fix-them-part-ii-article)
- [Common Motion Problems and How to Fix Them - Part III](https://www.pmdcorp.com/resources/type/articles/get/common-motion-problems-and-how-to-fix-them-part-iii-article)

mike4192, 2020-10-13:
Strange response from @EmperorKonstantin, it seems almost as if it is automatically generated or spam. In my opinion it is very generic and not very helpful.

@fjblau That's unfortunate, I'm sorry to hear your troubles.

I am indeed using the same servos and a 2S lipo battery, and I can't say I've experienced sluggish power output from them. I first used MG996R servos in my robot, and those were definitely way too weak and struggled to lift the robot similar to as you described.

Here's a random assortment of questions and tips:

- Are your servos connected directly to the Lipo (i.e.: not through a BEC)? Are you sure there are no voltage drops between it and the servos? Verify there isn't a voltage drop across the PCA9685 servo board if you are using that to connect the lipo to the servo power.
- Are you sure your lipo is fully charged?
- Is your robot frame excessively heavy? What capacity and C rating is your 2s lipo? Mine is 4000 mah and 25c I believe.
- Is there any excessive binding in your robot's joints? They should move relatively freely just rotating by hand (other than servo gearing resistance).
- Have you tried playing with the following config file values to shift the neutral stance position forward or back, effectively modifying the center of mass position? (stand_front_x_offset and stand_back_x_offset, these will effectively the robot's front and rear legs, respectively, forward or backwards in the neutral stance)

fjblau, 2020-10-15:
Thanks Mike!

I have made some progress but it is still a little jerky: https://www.youtube.com/watch?v=HjXhBpJFwJo

But at least she is moving forward!

mike4192, 2020-10-15:
Nice job!! From what I can see in the video, I would suggest trying to move your robot's center of mass forward (e.g: move the battery forward if you can). I think that might make it walk more stable.

fjblau, 2020-10-16:
I will try that... Thinking it could also be related to using the longer bed design...

EmperorKonstantin, 2020-10-22:
@mike4192  just giving the man general advice. There's some really good information in there about troubleshooting servo and motors problems in general. @fjblau is well on his way and I hope he accomplishes smooth motion soon.

fjblau, 2020-10-22:
Is there a way to slow down the "stand" command? I think that might help...

mike4192, 2020-10-22:
Yes, there are two relevant parameters in the config yaml file: transit_tau and transit_rl

These correspond to a transition time constant and transition rate limit, respectively. All state transitional movements go through a rate limited first order filter. Try decreasing the rate limit value from 0.6 (in units of meters/second), that will slow down the robot movement as it goes from sitting to standing.

fjblau, 2020-10-23:
Cool, I will try that... one other question... the PCA9685 board has a pre-installed 470uF capacitor... by my google-research, that is about 1/3 of the needed size for 12 servos. Has anyone tried swapping that out for a 1200uF and seen a difference in stability?

fjblau, 2020-10-23:
I lowered it to 0.4 and I am getting errors:

[ERROR] [1603487372.758163009]: Error setting PWM end high byte on servo 4 on board 1
[ERROR] [1603487372.758485045]: Error setting PWM start low byte on servo 5 on board 1
[ERROR] [1603487372.758862804]: Error setting PWM start high byte on servo 5 on board 1
[ERROR] [1603487430.878994970]: Error setting PWM end high byte on servo 5 on board 1
[ERROR] [1603487430.880866301]: Error setting PWM start high byte on servo 7 on board 1
[ERROR] [1603487430.881638522]: Error setting PWM end high byte on servo 7 on board 1
[ERROR] [1603487430.882261965]: Error setting PWM start high byte on servo 8 on board 1
[ERROR] [1603487430.882459984]: Error setting PWM end low byte on servo 8 on board 1
[ERROR] [1603487430.882741965]: Error setting PWM end high byte on servo 8 on board 1
[ERROR] [1603487430.882956372]: Error setting PWM start low byte on servo 9 on board 1
[ERROR] [1603487430.883233705]: Error setting PWM start high byte on servo 9 on board 1
[ERROR] [1603487430.893639062]: Error setting PWM start high byte on servo 1 on board 1
[ERROR] [1603487430.893841895]: Error setting PWM end low byte on servo 1 on board 1
[ERROR] [1603487430.895748318]: Error setting PWM start low byte on servo 3 on board 1

fjblau, 2020-10-23:
Also when I go to "stand" the position keeps moving in the spot_micro_motion_cmd.launch window:

[ WARN] [1603487479.875971418]: Joint RF_1, Angle: 0.00
[ WARN] [1603487479.894177580]: Proportional Command below -1.0 was computed, clipped to -1.0
[ WARN] [1603487479.894261598]: Joint RB_1, Angle: 0.00
[ WARN] [1603487479.894810653]: Proportional Command above +1.0 was computed, clipped to 1.0
[ WARN] [1603487479.895039042]: Joint RB_3, Angle: 99.12
[ WARN] [1603487479.895269097]: Proportional Command below -1.0 was computed, clipped to -1.0
[ WARN] [1603487479.895483689]: Joint RF_1, Angle: 0.00

But when I "idle" it does stop

mike4192, 2020-10-23:
Are you sure you don't inadvertently have two instances of some software running? The first errors I'm not very sure of, they are straight out of the the i2cpwn_board node. Maybe try power cycling everything. Changing the rate limit value shouldn't break things.

The second set of warning, what do you mean the position keeps moving? The warnings being printed mean joint angle commands beyond allowable range of motion are being computed for some reason.

fjblau, 2020-10-24:
Update on the Pi SpotMicro project... fixed the "Drunk Waschbär" gait problem and it seems to be walking a lot better!

https://www.youtube.com/watch?v=3bWrIb2TuO4

fjblau, 2020-10-26:
Also just out of curiosity... the default PWM frequency in the driver is 50hz for that servo. Is that the correct setting? I can't seem to find anything in the specs...

mike4192, 2020-10-27:
50 Hz is generally the standard frequency servo's PWM signals operate at. I would guess they still operate driven at different frequencies, but you probably can't squeeze any more performance out of them. The reason for the exact value is probably a holdover for some legacy reasons. See the following stack overflow post: https://electronics.stackexchange.com/a/397738

The c++ source code for the i2c pwm controller has a variable which can be used to modify this frequency (_pwm_frequency), but be warned, I have no idea what the effects may be or if it will even work when changed.

closed 2020-12-20 by mike4192