mike4192/spotMicro #18: somthing goes wrong with the body posture

leinaozhineng, 2020-12-01:
dear mike:
      thanks for your excellent work,but i have 2 problems after strict calibration.
      the 1st one is when the spotmicro is standing or walking, all the shoulders lean in the same direction,such like the picture.
[image]
      the 2nd problem is its right lower leg is always higher than other legs, so it doesn't provide force while standing and walking.
      i have no idea how to solve these problems after calibration for many times  even use a protractor following your documentation strictly.
     Any advice is much appreciated! And I really enjoy this project!

AAABBCND, 2020-12-01:
I have almost the same problem as you. I can only raise its right foot by sticking thick paper under it. However, this is not a fundamental solution to the problem and I sincerely hope to get Mike's answer.

mike4192, 2020-12-20:
Hi. Sorry to hear of your difficulties. Unfortunately, I don't know off the top of my head what your issue may be, but here are some random suggestions:
- Verify servo movement directions are correct. When you command the robot to do something, do all servos move in the correct direction?
- When you ran calibration, did you connect and disconnect some servos from their control horns? Did you accidentally reconnect a joint at a slight offset?
- Try manually rerunning the calibration procedure and filling in the spreadseet for a single joint (e.g. for right front hip). Then load up the python servo control node where you can control servos by keyboard. Go to the "Centerl PWM Val" and check if the joint looks like its at the "Calculated Angle at Center PWM Value". Calculate what the PWM value should be for 0 degrees (using the pwm/deg slope). Do similar for another angle. Basically, manually check the calibration on a single joint.

leinaozhineng, 2020-12-20:
thanks for your kind answer.finally，we solve these two problems by the following method: 1) follow the calibration procedure to callibrate all servos angle 2)command spot micro to stand up and then adjust the angle(the center value and sometimes the range ) according to the stand posture.
I know this is not a fundamental solution to the problem，but  spotMicro behaves very well in all the states, so i think this a good solution.

mike4192, 2020-12-20:
Glad you were able to figure it out. Will close the issue for now.

closed 2020-12-20 by mike4192