mike4192/spotMicro #24: Does this look like a servo config issue?

MZandtheRaspberryPi, 2020-12-22:
Hey Mike,

Thanks for the amazing software suite you've put together! It's incredible to be able to so quickly go from hardware to coordinated movements using it.

I'm having an issue where my spot when I ask it to stand is crooked and it looks like the servos are not configured correctly, though I did try to take care in measuring when doing the config. I was hoping to ask you for your thoughts! I'm using mostly the same hardware as you in terms of HV5523MG servos and a similiar servo board, but I do have an RPI4.

After I configure my servos by updating the .yaml file and run the below nodes,
```
rosrun i2cpwm_board i2cpwm_board
roslaunch spot_micro_motion_cmd spot_micro_motion_cmd.launch
rosrun spot_micro_keyboard_command spotMicroKeyboardMove.py
```
I then issue a stand command, my robot is tilted to the side and it looks like the upper servos in the "shoulders" are not going to the right position. Does this look like I've mis-calibrated the servos? I think not all the shoulder joints are in the center with equal movement to each side, but i kind of expected the software to compensate for that given it knows the center angle of each servo. Is that misguided?
[image]
[image]

I also see output from the i2cpwm_board node that indicates its calculating an angle outside of the movement range of the servo and clipping it, which would seem to indicate something about my servo config is suspect. The below is a sample:
```
[ WARN] [1455213262.147691540]: Joint LB_2, Angle: 40.64
[ WARN] [1455213262.148428929]: Proportional Command below -1.0 was computed, clipped to -1.0
[ WARN] [1455213262.148982948]: Joint LB_3, Angle: -126.58
[ WARN] [1455213262.149471059]: Proportional Command below -1.0 was computed, clipped to -1.0
[ WARN] [1455213262.150248114]: Joint LF_3, Angle: -125.66
[ WARN] [1455213262.150982892]: Proportional Command above +1.0 was computed, clipped to 1.0
[ WARN] [1455213262.151215244]: Joint RB_3, Angle: 126.58
[ WARN] [1455213262.151978096]: Proportional Command above +1.0 was computed, clipped to 1.0
[ WARN] [1455213262.152449337]: Joint RF_3, Angle: 125.66
[ WARN] [1455213262.167402744]: Proportional Command above +1.0 was computed, clipped to 1.0
```

This is what the spreadsheet to calculate config values looks like, and the same values are input into the .yaml config file. Some of the values look a tad odd, and different from the values in your spreadsheet in the repo, in particular the configuration for the upper shoulders where i configured to -45 and the repo spreadsheet configures to -90 as angle 2, for example looking at Servo 2.
[image]

I can try taking it apart and re-centering servos, but I also wanted to ask if you had any ideas for me to take a first look at. I appreciate your thoughts!

MZandtheRaspberryPi, 2020-12-22:
Oh, I think I am possibly mixing up the hip and shoulders with the upper leg in the config. Does the hip refer to the servos embedded in the head/butt of the robot, right next to the chasis/body? Ie the highlighted part in the screenshot below?

[image]

Then the shoulder is also called the upper leg? That could explain my confusion and then I definitely misconfigured my servos...

MZandtheRaspberryPi, 2020-12-23:
I've actually corrected that and its much better now though I seem to have a servo or two that is "bad" and doesn't move all the way to commanded positions sometimes which makes standing up harder. I'll see about debugging that though, it's exciting to see it stand even if shakily!

closed 2020-12-23 by MZandtheRaspberryPi

mike4192, 2020-12-23:
Hey sorry for the delay getting back to you. Have you resolved your issues? You mentioned you still have a servo or two is bad.

To answer your earlier questions, yes, the hips refers to the inner most joints attached to the robot frame. Lower leg is the knee joint, and upper leg is the joint after the knee.

MZandtheRaspberryPi, 2020-12-29:
Thank you for the confirmation!

I just resolved the issues. I think I screwed some servo horns too tightly in. When the servo was moving through its range of motion, there were a couple places where it shook quite badly and couldn't get to a location. When I took it apart, put it back together, and made sure screws weren't too tight it worked much better. I actually got it walking just now which is pretty miraculous to me. I'll post a video a little later.

Thanks for all the support!

MZandtheRaspberryPi, 2021-01-02:
Here's a video, thanks for all the help: https://www.youtube.com/watch?v=g6FMSfrcahk

I have some other stuff to do like some network issues with the RPI4, and i haven't installed any screens yet, but walking was my first goal, so I'm happy to be here. Thanks again for the awesome software!

mike4192, 2021-01-03:
Awesome work man!! 

I know it took a lot of work to get to this point. I'm glad my worked helped you get your robot up and running.