mike4192/spotMicro #26: Not able to compile on Ubuntu 20.04.1

gio1993, 2020-12-27:
Hi, 
I've cloned the project int my catkin_ws. I'm running Ubuntu 20.04 and ROS noetic. I've also downloaded the libi2c-dev_3.1.1-1.deb package and installed it. When I try compiling my catkin_ws I get a set of errors due to undefined references. Here's my log:

`giovanni@Giovanni-HP:~/catkin_ws$ catkin_make -DCMAKE_BUILD_TYPE=Release
Base path: /home/giovanni/catkin_ws
Source space: /home/giovanni/catkin_ws/src
Build space: /home/giovanni/catkin_ws/build
Devel space: /home/giovanni/catkin_ws/devel
Install space: /home/giovanni/catkin_ws/install
####
#### Running command: "cmake /home/giovanni/catkin_ws/src -DCMAKE_BUILD_TYPE=Release -DCATKIN_DEVEL_PREFIX=/home/giovanni/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/giovanni/catkin_ws/install -G Unix Makefiles" in "/home/giovanni/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/giovanni/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/giovanni/catkin_ws/devel;/opt/ros/noetic
-- This workspace overlays: /home/giovanni/catkin_ws/devel;/opt/ros/noetic
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.8.5", minimum required is "3") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Using Debian Python package layout
-- Using empy: /usr/lib/python3/dist-packages/em.py
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/giovanni/catkin_ws/build/test_results
-- Forcing gtest/gmock from source, though one was otherwise available.
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python3 (found version "3.8.5") 
-- Using Python nosetests: /usr/bin/nosetests3
-- catkin 0.8.9
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 11 packages in topological order:
-- ~~  - manipulator_moveit_config
-- ~~  - i2cpwm_board
-- ~~  - lcd_monitor
-- ~~  - servo_move_keyboard
-- ~~  - spot_micro_keyboard_command
-- ~~  - spot_micro_motion_cmd
-- ~~  - spot_micro_plot
-- ~~  - spot_micro_simple_command
-- ~~  - spot_micro_walk
-- ~~  - manipulator_description
-- ~~  - moveit_converter
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'manipulator_moveit_config'
-- ==> add_subdirectory(manipulator_moveit_config)
-- +++ processing catkin package: 'i2cpwm_board'
-- ==> add_subdirectory(spotMicro/ros-i2cpwmboard)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- i2cpwm_board: 6 messages, 4 services
-- +++ processing catkin package: 'lcd_monitor'
-- ==> add_subdirectory(spotMicro/lcd_monitor)
-- +++ processing catkin package: 'servo_move_keyboard'
-- ==> add_subdirectory(spotMicro/servo_move_keyboard)
-- +++ processing catkin package: 'spot_micro_keyboard_command'
-- ==> add_subdirectory(spotMicro/spot_micro_keyboard_command)
-- +++ processing catkin package: 'spot_micro_motion_cmd'
-- ==> add_subdirectory(spotMicro/spot_micro_motion_cmd)
-- +++ processing catkin package: 'spot_micro_plot'
-- ==> add_subdirectory(spotMicro/spot_micro_plot)
-- +++ processing catkin package: 'spot_micro_simple_command'
-- ==> add_subdirectory(spotMicro/spot_micro_simple_command)
-- +++ processing catkin package: 'spot_micro_walk'
-- ==> add_subdirectory(spotMicro/spot_micro_walk)
-- +++ processing catkin package: 'manipulator_description'
-- ==> add_subdirectory(manipulator_description)
-- +++ processing catkin package: 'moveit_converter'
-- ==> add_subdirectory(moveit_converter)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- moveit_converter: 1 messages, 0 services
[... 178 lines cut ...]
/usr/bin/ld: CMakeFiles/test_spot_micro_leg.dir/tests/test_spot_micro_leg.cpp.o:(.data.rel.ro._ZTI34setFootPos_global_coordinates_Test[_ZTI34setFootPos_global_coordinates_Test]+0x10): undefined reference to `typeinfo for testing::Test'
/usr/bin/ld: CMakeFiles/test_spot_micro_leg.dir/tests/test_spot_micro_leg.cpp.o:(.data.rel.ro._ZTV33setFootPos_local_coordinates_Test[_ZTV33setFootPos_local_coordinates_Test]+0x20): undefined reference to `testing::Test::SetUp()'
/usr/bin/ld: CMakeFiles/test_spot_micro_leg.dir/tests/test_spot_micro_leg.cpp.o:(.data.rel.ro._ZTV33setFootPos_local_coordinates_Test[_ZTV33setFootPos_local_coordinates_Test]+0x28): undefined reference to `testing::Test::TearDown()'
/usr/bin/ld: CMakeFiles/test_spot_micro_leg.dir/tests/test_spot_micro_leg.cpp.o:(.data.rel.ro._ZTV34setFootPos_global_coordinates_Test[_ZTV34setFootPos_global_coordinates_Test]+0x20): undefined reference to `testing::Test::SetUp()'
/usr/bin/ld: CMakeFiles/test_spot_micro_leg.dir/tests/test_spot_micro_leg.cpp.o:(.data.rel.ro._ZTV34setFootPos_global_coordinates_Test[_ZTV34setFootPos_global_coordinates_Test]+0x28): undefined reference to `testing::Test::TearDown()'
/usr/bin/ld: ../../../../lib/libgtest_main.so: undefined reference to `testing::UnitTest::GetInstance()'
/usr/bin/ld: ../../../../lib/libgtest_main.so: undefined reference to `testing::UnitTest::Run()'
/usr/bin/ld: ../../../../lib/libgtest_main.so: undefined reference to `testing::InitGoogleTest(int*, char**)'
collect2: error: ld returned 1 exit status
make[2]: *** [spotMicro/spot_micro_motion_cmd/libs/spot_micro_kinematics_cpp/CMakeFiles/test_spot_micro_leg.dir/build.make:87: spotMicro/spot_micro_motion_cmd/libs/spot_micro_kinematics_cpp/test_spot_micro_leg] Errore 1
make[1]: *** [CMakeFiles/Makefile2:2042: spotMicro/spot_micro_motion_cmd/libs/spot_micro_kinematics_cpp/CMakeFiles/test_spot_micro_leg.dir/all] Errore 2
make[1]: *** Attesa per i processi non terminati....
[ 84%] Building CXX object spotMicro/spot_micro_motion_cmd/CMakeFiles/smfsm.dir/src/smfsm/spot_micro_stand.cpp.o
[ 86%] Building CXX object spotMicro/spot_micro_motion_cmd/CMakeFiles/smfsm.dir/src/smfsm/spot_micro_transition_idle.cpp.o
[ 86%] Building CXX object spotMicro/spot_micro_motion_cmd/CMakeFiles/smfsm.dir/src/smfsm/spot_micro_transition_stand.cpp.o
[ 87%] Building CXX object spotMicro/spot_micro_motion_cmd/CMakeFiles/smfsm.dir/src/smfsm/spot_micro_walk.cpp.o
[ 88%] Linking CXX executable /home/giovanni/catkin_ws/devel/lib/i2cpwm_board/i2cpwm_board
[ 88%] Built target i2cpwm_board
[ 90%] Linking CXX shared library /home/giovanni/catkin_ws/devel/lib/libsmfsm.so
[ 90%] Built target smfsm
make: *** [Makefile:141: all] Errore 2
Invoking "make -j4 -l4" failed
`
Any idea of what the issue can it be?
Regards.

mike4192, 2020-12-27:
Unfortunately ROS Kinetic, which this project is based on, is only supported in Ubuntu 16.04, so not sure my suggestions will help at all, but it looks like you may be missing gtest.

Some random suggestions:

- Make sure you have gtest installed: `sudo apt-get install libgtest-dev`
- Make sure you initialized all the submodules: 
```
git submodule update --init --recursive
git submodule update --recursive
```

gio1993, 2020-12-28:
Thank you for your suggestion but libgtest-dev was already installed and updated.
I reinstalled ROS Noetic, followed your instructions and after some real struggling with python versions I got it compiling and running (on an old Laptop for now). During next days I'll implement it on RPI4 and I'll try it on Ubuntu 20.1. In case i'll not be able to make it run, I'll try the 20.04 release. 
I'll let you know if you are interested.
Regards.

closed 2020-12-28 by gio1993