mike4192/spotMicro #40: Add Adafruit 9DOF BNO055 IMU

mike4192, 2021-01-09:
This will be hard because hardware i2c can't be used due to a RPi hardware limitation regarding i2c clock stretching.

Possible options:

- Use software i2c: undesirable because it will eat up CPU
- Figure out how to use UART: best bet, but most guides online are for configuring raspian to do this. Need to figure out how to do it on ubuntu
- Use an arduino middle man, and communicate over USB via serial: not ideal, but if other options fall through it is a solution. Use arduino nano

sbias, 2021-02-26:
Just a note: the "new" firmware of raspberry should be able to do some clock stretching, but i down know if this conflicts with the PCA9685.

mike4192, 2021-02-27:
What firmware are you referring to? I believe it's a hardware bug and there are no fundamental fixes, only workarounds.

See: https://www.recantha.co.uk/blog/?p=19880

It seems it even on the RPi4 still.

sbias, 2021-02-28:
TBH – i can not find that article i read anymore
since i don't needed it yet ,i havn't bookmarked it. But iirc they did nothing else than that in the article you linked above. As far as i can see (now) its just slowing down the whole bus, not stretching it if needed. So you might right, seems like uart is the best solution for it.
Maybe i should get one and give it a go. Since it seems to not be currently available by adafruit, i will try another breakout-board with a BNO055. By reading the docs its just important to get the ps0 and ps1 to configure it to uart. (?)

sbias, 2021-03-09:
Okey, i tried it the uart way based on the image you recommend in your readme.
Was pretty easy:

Configure the pi to enable uart:
sudo raspi-config
 -> interface options
  -> serial
   -> yes
reboot

Configure the BNO055 to uart, that is a hardware-config only. see https://github.com/RoboticArts/ros_imu_bno055#2-hardware-setup how to do it. For another board just follow the configuration for PS0 = 1 and PS1 = 0, connect it to your pi. I used the 3.3V pin for power. The BNO055 works fine with that voltage and i need any level-shifters :)

follow https://github.com/RoboticArts/ros_imu_bno055#3-dependences and https://github.com/RoboticArts/ros_imu_bno055#4-installation

edit **ros_imu_bno055/launch/imu.launch** and change serialport to **/dev/ttyAMA0**

**run roslaunch ros_imu_bno055 imu.launch**

with **rostopic echo /imu/data** you should see:
---
```
header: 
  seq: 773
  stamp: 
    secs: 1615309390
    nsecs: 977561235
  frame_id: "imu_link"
orientation: 
  x: 12160.0
  y: -9219.0
  z: -1730.0
  w: 5708.0
orientation_covariance: [-1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity: 
  x: 0.01
  y: 0.02
  z: -0.00111111111111
angular_velocity_covariance: [-1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
linear_acceleration: 
  x: 0.05
  y: 0.06
  z: 0.11
linear_acceleration_covariance: [-1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]

```

now, you need to define a position where to place it. i would say the best position is the exact center. so the height oh the hip-servo-axes and in x/y direction the exact center. this should prevent acceleration if the robot is just rotating ...

mike4192, 2021-03-10:
Nicely done. I may try this out this weekend.

Originally, I thought that since I was running an Ubuntu 16.04 [Ubiquity Robotics Image](https://downloads.ubiquityrobotics.com/pi.html) on my RPi, it wouldn't have `raspi-config`, but it turns out it does!

Apparently this was preinstalled in the image already. I thought the script was inherrent to Raspbian OS only, but looks like it was at least ported to Ubuntu 16.04 at some point.