mike4192/spotMicro #43: Servo installation angles

niranjan1997, 2021-01-29:
Hey Mike, I am using mg996r servos in my robot. I have followed the calibration instructions given by you. But what are the angles between shoulder and knee links when the servos are at their neutral stance ?

mike4192, 2021-01-29:
I'm not sure I fully understand your question. Are you asking about the knee angle (angle of the lower leg relative to the upper leg)? And are you asking about what that angle should be during installation, when the servo is powered on and in it's neutral position? Or are you asking about what that knee angle is with the robot in a standing stance?

niranjan1997, 2021-01-30:
I am asking about what that angle should be during installation when the servo is powered on and in its neutral position? and what is the knee and shoulder angle is with the robot in a standing stance?

mike4192, 2021-01-30:
I'm not sure exactly what the knee angle is in a standing stance, but I'd say it is approximately 90 degrees. I installed the servos such that the legs were oriented approximately as shown in the image below (from the calibration guide). Honestly I just eyeballed it.

The important thing is trying to maintain consistency installing servos in and calibrating all four legs.

[image]

mike4192, 2021-02-22:
Gonna close this unless you have further questions.

closed 2021-02-22 by mike4192