mike4192/spotMicro #44: RLException:  while Running roslaunch

niranjan1997, 2021-01-29:
I am getting this error while running the command `roslaunch spot_micro_lauch motion_control_and_hector_slam.launch`

RLException: [motion_control_and_hector_slam.launch] is neither a launch file in package [spot_micro_lauch] nor is [spot_micro_lauch] a launch file name
The traceback for the exception was written to the log file
the build was successful. without any errors. 
I have sourced my workspace. using source devel/setup.bash.
please help me in resolving this issue.

mike4192, 2021-01-29:
Are you using catkin build? See the following page for info on setting up a workspace correctly:
https://catkin-tools.readthedocs.io/en/latest/quick_start.html

I just created a test catkin workspace from scratch and checked out my repo and built it and was able to launch without issue.
```
My commands were:
source /opt/ros/kinetic/setup.bash 
mkdir -p ~/catkin_temp/src            
cd ~/catkin_temp       
catkin init                                
cd ~/catkin_temp/src      
git clone https://github.com/mike4192/spotMicro.git .
git submodule init
git submodule update
cd ..
catkin build
source devel/setup.bash
roslaunch spot_micro_lauch motion_control_and_hector_slam.launch
```

niranjan1997, 2021-01-30:
[image]
I have followed the above instructions. created a temp catkin ws cloned the repo and built it with catkin build command. even after a successful build. when I try to launch the file I am getting the same error.
`RLException: [motion_control_and_hector_slam.launch] is neither a launch file in package [spot_micro_lauch] nor is [spot_micro_lauch] a launch file name
The traceback for the exception was written to the log file
`

mike4192, 2021-01-30:
You have a typo in your command. You are putting `lauch` instead of `launch`

niranjan1997, 2021-01-30:
Yeah, I just figured it out.  thank you. :D

closed 2021-01-30 by niranjan1997

mike4192, 2021-01-30:
Thank you. Will update it now