mike4192/spotMicro #49: A bit of uncertainty in servo_calibration ods

sbias, 2021-02-19:
First words: i'm unsure i this is an error in that ods or my failure. After all, my spotmicro is walking okay, first try.
But, that ods to calculate the servo calibration values was a bit irritating for 2 lines - which looks a bit like they where swapped.
If i compare the values of hip-servos 3 and 12 this the values from this documentation, they are way of. In fact even the direction is the opposite. The other offsets are in a reasonable range. I've attached a screenshot of my values.

After rechecking that multiple times i gave it a go and started everything. micro is walking quite good with this values.

[image]

mike4192, 2021-02-22:
It definitely seem as if your directions for those servos are somehow reversed from what I have. I'm trying to think how this could be possible but I'm kinda stumped. You're the first to report something like this I don't think it's an error in the spreadsheet or calibration procedures.

Servos 3 and 12 are the servos for the front two hip joints. 
- Are you sure the connections for these servos are not accidentally swapped?
- How did you mount these servos? Is their orientation is somehow flipped? Is it possible you mounted the hip joints upside down somehow in the front? (though, I'm not sure if it's possible to mount them in a different orientation, or if it would cause this discrepancy)

Two hip servos on the same side of the robot (e.g. the right side) are oriented 180 deg relative to each other about the Z axis (in the TF2 coordinate frame, see diagrams in additional_hardware_description.md). Because of this, a rotation of each servo in their own coordinate frames (i.e. in the direction protruding out of the servo control horn) results in opposite leg motion. For example, considering the right side, a clockwise rotation of the front right hip servo in it's coordinate frame results in leg rotation downwards, while a clockwise rotation of the back right servo in it's frame results in a leg rotation upwards. Thus, in my calibration spreadsheet the front and back servos each have opposite rotational directions, whereas in your spreadsheet, the front and back servos on a given left or right side both have the same rotational direction, which seems impossible.

sbias, 2021-02-22:
Thanks for your explanation, i think that helped a lot to find the problem. After rethinking everything you wrote vs everything i did while building, i found a difference:
My hip servos  are all pointing backwards. I think i printed the "wrong" model for this repo: https://www.thingiverse.com/thing:4155673 (you can find my make there too)
If the servos are all pointing outwards in the original and on that model all backwards, that will explain the issue. That change wasn't mentioned there. If you can confirm my theory, i will poke im to write a hint. (Awesome the procedure still works :) )
Maybe i should measure all links-length to confirm they are correct since walking isn't that stable yet.

mike4192, 2021-02-24:
Yes I think your right. If both hip servos on the left side of your robot, for instance, are oriented the same direction, then that would explain the configuration value differences.

I can add a note to the readme about this, as well as link to this thingverse model. I'm all in favor of using an easier model as the original is pretty tricky to assemble.

nahueltaibo, 2021-02-25:
Hi guys, 
   As far as I remember, I got based on a previous version of spot micro to position those inverted servos. And I was not aware of this code repo until @sbias mentioned it in a comment on thingiverse. Its nice to see the robot moving! I modified the design, printed it but never got to coding it in the end

I'd like to improve the description on [thingiverse ](https://www.thingiverse.com/thing:4155673)so other people are alerted about this servos being inverted.
What do you think I should mention in there?

sbias, 2021-02-25:
I think it should be enough to mention it along the list of improvements you made to the model. At least for this software everything works just fine, just spent some additional time since the values i got wasn't logically explainable. I know there is some more software out there to run on spotmicro, unsure if that might be a bigger problem there. So a hint to that would be awesome.
But, if i got you right, this servo-position came from an older version of the remixed model? If true, might be a good hint in this documentation for servo-calibration too.

mike4192, 2021-02-27:
+1 on adding a bullet to the improvement section that the servos were reoriented compared to KDY's design.

PR #53 (2021-03-05): add hint for inverted hip servos to servo calibration documentation
Add hint for inverted hip servos to calibration procedure
Closes #49
closed 2021-03-05 by mike4192