mike4192/spotMicro #59: UBUNTU 20.4 / ROS NOETIC - LTS VERSION

jpereirajr, 2021-03-18:
I have worked in a version for UBUNTU 20.4 and ROS NOETIC, both of them LTS version, with support until 2025. 
Furthermore, NOETIC uses python 3 by default, once python 2.7 is deprecated.

It is 100% functional,  resulting faster and with long term support, also it will make much easer migrate to ROS 2, if it is the choice.

Instead create a fork to work in it, if you agree, I can share all changes and the final code, which we could mainting as collaborators.

mike4192, 2021-03-20:
Hi.

Sure you can create a fork and I can take a look to see what changes are necessary for this project to work in ubuntu 20.04.

jpereirajr, 2021-03-29:
Hi

The fork is published.
https://github.com/jpereirajr/spotMicro

Take a look and let me know

mike4192, 2021-04-02:
Thanks! Nice work. I will take a look eventually, but don't expect a fast turnaround time.

mchobbylong, 2021-04-24:
**GREAT to know that someone is working on migrating to newer version of ROS!** Thx for your work.

I am trying to build this repo on RPi4B + Ubuntu 20.04 + ROS noetic too. But I just found that the package `ros-i2cpwmboard` is not working correctly...

Below is the log from `rosrun i2cpwm_board i2cpwm_board`:

```
[ INFO] [1619295376.431822812]: I2C bus opened on /dev/i2c-1
[ INFO] [1619295376.446875080]: Setting PWM frequency to 50 Hz
[FATAL] [1619295377.457911782]: Failed to open I2C bus (null)
[ INFO] [1619295377.460355694]: Setting PWM frequency to 50 Hz
[1]    11746 segmentation fault (core dumped)  rosrun i2cpwm_board i2cpwm_board
```

After some time for investigation, I believe the package somehow runs twice, while it failes on the second time. There is also another one complaining this issue in the [package's original repo](https://gitlab.com/bradanlane/ros-i2cpwmboard/-/issues/8).

@jpereirajr Did you encounter this issue? If so, then how do you solve it?

jpereirajr, 2021-05-01:
> **GREAT to know that someone is working on migrating to newer version of ROS!** Thx for your work.
>
> I am trying to build this repo on RPi4B + Ubuntu 20.04 + ROS noetic too. But I just found that the [...]

**Hi mchobbylong.**

**I couple of question, to try help you.**
1. What version of ubuntu are you using, 64 or 32 bits?
2. Have you recompile the code from the begin?
3. Was the compilation succeful?

**A couple of possible checks:**

1. I got some error on the i2cpwm_board package that I resolved with two changes:
     - I added `i2c` into line 22 of `src/ros-i2cpwmboard/CMakeLists.txt`, so that it looks like:
```target_link_libraries(i2cpwm_board ${catkin_LIBRARIES} i2c)```

     - And  I added the code below to `src/ros-i2cpwmboard/src/i2cpwm_controller.cpp`:
```
extern "C" {
	#include <i2c/smbus.h>
}
```
2. What is the out put of `i2cdetect -y 0` and `i2cdetect -y 1`? Expected
```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: 40 -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: 70 -- -- -- -- -- -- --   
```
3. Check if you have the i2c enabled in `/boot/firmware/syscfg.txt`
```
dtparam=i2c1=on
dtparam=i2c_arm=on
dtparam=spi=on
```
4. Check if the `ubuntu` user is configured in the group i2c
`/etc/group`
```
 i2c:x:119:ubuntu
```

mchobbylong, 2021-05-06:
@jpereirajr Good to hear from you again! And thanks for your diagnosis!

> What version of ubuntu are you using, 64 or 32 bits?
> Have you recompile the code from the begin?
> Was the compilation succeful?

I was using Ubuntu Server 20.04 32 bits. I re-compiled the code with needed libraries (i2cpwm-board, hector-slam, etc.) from the beginning (well, if I didn't misunderstand it, cloning the repo to `src` folder and `catkin init && catkin build` should be a compilation from "beginning"). And of course the compilation was successful except for a few warnings came from the package `i2cpwm-board`.

> I got some error on the i2cpwm_board package that I resolved with two changes

Seems that someone have already fixed this with the latest commits from the original repo. I merged them using `cd ros-i2cpwmboard && git pull`, then the issue you mentioned was solved.

> What is the out put of `i2cdetect -y 0` and `i2cdetect -y 1`?

`i2cdetect -y 1` gave me nearly the same result as yours, except that since I've also attached a LCD screen in parallel, port 27 was also available.

> Check if you have the i2c enabled in `/boot/firmware/syscfg.txt`

**This could be the key to the puzzle, since I don't remember that I have made modifications to that file, but only installed `libi2c-dev` via apt. I'll check that.**

> Check if the ubuntu user is configured in the group i2c

Yes, I had already added `ubuntu` into group i2c via `sudo useradd -g i2c ubuntu`.

---

**Unfortunately, since I have switched to Ubuntu 18.04 + ROS Melodic just for catching deadlines, guess I have to redo all the things from scratch. I will give feedbacks to you later.**

jpereirajr, 2021-05-06:
Hi @mchobbylong.

Only an additional question, you are using the source from my fork, right?

I am current using ubuntu 64 bits server version. I am deploying a 32 bits to test it, once now it is already support and have noetic packages, and soon I comment here.

Only a change, the hector-slam, now is in the Ubuntu reposiboty.

mchobbylong, 2021-05-07:
> Only an additional question, you are using the source from my fork, right?

Absolutely yes.

Maybe Ubuntu 64bits has better support, not sure about that... I will also try building your fork on Ubuntu Server 20.04 64bits later.

modulo1, 2021-05-09:
I wanted to add to this, as I had just got my spotmicro to the point where I can test the software.

I installed the noetic fork on Ubuntu 20.04 Server 64-bit (using the ubuntu-20.04.2-preinstalled-server-arm64+raspi.img.xz image) on a Raspberry Pi 4B (4GB), using ros-noetic-desktop-full and the other dependencies needed.  After making the necessary changes to CMakelists.txt and i2cpwm_controller.cpp [seen here](https://github.com/mike4192/spotMicro/issues/59#issuecomment-830647917). I tried running the keyboard control package, and got the error and segfault described [here](https://github.com/mike4192/spotMicro/issues/59#issuecomment-826148739). 

I then used a 64-bit Raspbian Buster image on the same Pi (using the 2021-03-04-raspios-buster-arm64.zip image).  Installed ros-noetic-desktop (only an 8GB microSD card so space was tight this time) and dependencies, added fixes to CMakelists.txt and i2cpwm_controller.cpp as above, and compiled. This time, I started with `roslaunch spot_micro_motion_cmd motion_cmd.launch` and was able to get the results of my controller axes moving in the terminal window.

Also, notably, no errors from the i2cpwm_board node:

>process[i2cpwm_board_node-2]: started with pid [2792]
process[spot_micro_motion_cmd_node-3]: started with pid [2793]
[ INFO] [1620499359.141873780]: I2C bus opened on /dev/i2c-1
from Constructor
[ INFO] [1620499359.161777315]: Setting PWM frequency to 50 Hz
[ INFO] [1620499360.171361530]: Servo #9 configured: center=306, range=374, direction=-1
[ INFO] [1620499360.171515491]: Servo #8 configured: center=306, range=392, direction=1
[ INFO] [1620499360.171560731]: Servo #7 configured: center=306, range=390, direction=1
[ INFO] [1620499360.171601379]: Servo #12 configured: center=306, range=389, direction=1
[ INFO] [1620499360.171642378]: Servo #11 configured: center=306, range=397, direction=1
[ INFO] [1620499360.171696822]: Servo #10 configured: center=306, range=387, direction=1
[ INFO] [1620499360.171741303]: Servo #6 configured: center=306, range=411, direction=1
[ INFO] [1620499360.171780135]: Servo #5 configured: center=306, range=396, direction=1
[ INFO] [1620499360.171817320]: Servo #4 configured: center=306, range=396, direction=1
[ INFO] [1620499360.171855320]: Servo #3 configured: center=306, range=396, direction=-1
[ INFO] [1620499360.171892264]: Servo #2 configured: center=306, range=389, direction=1
[ INFO] [1620499360.171928763]: Servo #1 configured: center=306, range=372, direction=1

I hadn't made any changes to spot_micro_motion_cmd.yaml yes, so the servo values are all over the place.

If I had to hazard a guess, there is still something missing from the i2c configuration that you get in Ubuntu.  I know the default 'ubuntu' user is a member of the 'dialout' group, which I velieve is necessary for i2c communication, and I enabled i2c in /boot/firmware/syscfg.txt.

`sudo i2cdetect -y 1` showed the pwm controller on 0x40, as well as an LCD screen at 0x27 and a GY-521 module at 0x68, so I'm fairly confident i2c was working.

[This](https://askubuntu.com/a/1273900) StackOverflow post had some instructions about creating an i2c group, and chown-ing the i2c tree in /dev. I tried that, but was still getting the segfault on Ubuntu.

Hopefully this helps some people. I'm still very new to ROS.

DennisGarvey, 2021-05-12:
> > **GREAT to know that someone is working on migrating to newer version of ROS!** Thx for your work.
> > I am trying to build this repo on RPi4B + Ubuntu 20.04 + ROS noetic too. But I just found that the package `ros-i2cpwmboard` is not working correctly...
> > Below is the log from `rosrun i2cpwm_board i2cpwm_board`:
> > ```
> > [ INFO] [1619295376.431822812]: I2C bus opened on /dev/i2c-1
> > [ INFO] [1619295376.446875080]: Setting PWM frequency to 50 Hz
> > [FATAL] [1619295377.457911782]: Failed to open I2C bus (null)
> > [ INFO] [1619295377.460355694]: Setting PWM frequency to 50 Hz
> > [1]    11746 segmentation fault (core dumped)  rosrun i2cpwm_board i2cpwm_board
> > ```
> > 
> > 
> > After some time for investigation, I believe the package somehow runs twice, while it failes on the second time. There is also another one complaining this issue in the [package's original repo](https://gitlab.com/bradanlane/ros-i2cpwmboard/-/issues/8).
> > @jpereirajr Did you encounter this issue? If so, then how do you solve it?
> 
> **Hi @mchobbylong.**
> 
> **I couple of questions, to try help you.**
> 
> 1. What version of ubuntu are you using, 64 or 32 bits?
> 2. Have you recompile the code from the begin?
> 3. Was the compilation succeful?
> 
> **A couple of possible checks:**
> 
> 1. I got some error on the i2cpwm_board package that I resolved with two changes:
>    
>    * I added `i2c` into line 22 of `src/ros-i2cpwmboard/CMakeLists.txt`, so that it looks like:
>      `target_link_libraries(i2cpwm_board ${catkin_LIBRARIES} i2c)`
>    * And  I added the code below to `src/ros-i2cpwmboard/src/i2cpwm_controller.cpp`:
> 
> ```
> extern "C" {
> 	#include <i2c/smbus.h>
> }
> ```
> 
> 1. What is the out put of `i2cdetect -y 0` and `i2cdetect -y 1`? Expected
> 
> ```
>      0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
> 00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
> 10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
> 20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
> 30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
> 40: 40 -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
> 50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
> 60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
> 70: 70 -- -- -- -- -- -- --   
> ```
> 
> 1. Check if you have the i2c enabled in `/boot/firmware/syscfg.txt`
> 
> ```
> dtparam=i2c1=on
> dtparam=i2c_arm=on
> dtparam=spi=on
> ```
> 
> 1. Check if the `ubuntu` user is configured in the group i2c
>    `/etc/group`
> 
> ```
>  i2c:x:119:ubuntu
> ```

I'm also having trouble with the i2cpwmboard node. I have tried all the steps you have provided and I am still getting the "Failed to open I2C bus /dev/i2c-1" error.

mchobbylong, 2021-05-17:
@jpereirajr @modulo1 @DennisGarvey

Good news here. I reinstalled Ubuntu Server 20.04.2 64bits on RPi4B, then just install all the softwares needed and compile the workspace **without upgrading any package manually through `sudo apt upgrade`**, and now I got no error from the command `rosrun i2cpwm_board i2cpwm_board`.

I guess `sudo apt upgrade` may be the key, since before this run I ALWAYS do `sudo apt full-upgrade` on a newly installed OS. Upgrading kernel modules may break the I2C support eventually.

But till now my robotdog does not stand up... Looks like there are other configuration problems in the I2C node itself, since I can now publish servo values to the I2C node, but the node doesn't echo those values and the servos are not moving.

I will try to figure out what's wrong in the next few days...

jpereirajr, 2021-05-18:
Hi @mchobbylong
I am sorry my delay, it has been totally busy days.
I just install a new ubuntu server from zero to replicate the full process.

On the next days  I will give you  a feedback

modulo1, 2021-05-18:
@mchobbylong 

This may or may not be related to your issues, but my spotmicro is using PDI-HV5523MG servos, and they would barely move using the servoMoveKeyboard.py ros package.  Actually, they would move about 90 degress as PWM value increased, and then reset and rotate back to where they started.

I had to change `_pwm_frequency` on line 283 of i2cpwm_controller.cpp to something a lot higher to get the servos to move anywhere past 90 degrees and stay.  Keep in mind if you change this value, you'll need to redo the servo calibration values in  [spot_micro_motion_cmd.yaml](https://github.com/mike4192/spotMicro/blob/master/spot_micro_motion_cmd/config/spot_micro_motion_cmd.yaml). I definitely did.

mike4192, 2021-05-18:
Sorry I haven't really been helping out on this. I haven't been working on the project recently, let alone migrating to newer Ubuntu.

I did want to jump in and express concern about your comment for changing the servo frequency. There's no reason the servo should not operate at 50hz pulsing frequency, so just warning you that this may be indicative of some deeper issue with the i2c interface and newer Ubuntu and the i2cpwm board package.

modulo1, 2021-05-18:
> 
> 
> Sorry I haven't really been helping out on this. I haven't been working on the project recently, let alone migrating to newer Ubuntu.
> 
> I did want to jump in and express concern about your comment for changing the servo frequency. There's no reason the servo should not operate at 50hz pulsing frequency, so just warning you that this may be indicative of some deeper issue with the i2c interface and newer Ubuntu and the i2cpwm board package.
> 
> Just be forewarned you may only be putting bandaids around symptoms and not addressing root causes.

I just posted a similar message on the spotmicro discord asking a similar question, as have I started to doubt myself after doing some further reading.

I was using Fran Ferri's calibration script from [here](https://gitlab.com/custom_robots/spotmicroai/basic-runtime/-/blob/master/calibration/calibration/calibration.py), which uses 50 for the PWM frequency.  I was able to get full 180 degree motion using that script. However, upon using the joystick movement, would crash as it could not find a second PCA9685 board, despite only one defined in my configuration file.

My inexperienced mind tells me it might be related to the clock speed the board is working at ("reference_clock_speed": 25000000 in Fran Ferri's script) but I didn't see anywhere clock speed was referenced in `i2cpwm_controller.cpp`.  I just can't think of another reason why there is such a discrepancy in performance beween Fran Ferri's implementaion, and the excellent work that you and @jpereirajr have done.

jpereirajr, 2021-05-18:
> I just posted a similar message on the spotmicro discord asking a similar question, as have I started to doubt myself after doing some further reading.
>
> I was using Fran Ferri's calibration script [...]

Hi @modulo1

It is working with 50hz. Some time, the issue could be a power source issue, that some times change of the frequency maskared the issue. Check if your servos are receiving the right current.

@mike4192, using the topic, how are you powering your servos?

jpereirajr, 2021-05-18:
> Hi @mchobbylong
> I am sorry my delay, it has been totally busy days.
> I just install a new ubuntu server from zero to replicate the full process.
> 
> On the next days I will give you a feedback

> @jpereirajr @modulo1 @DennisGarvey
>
> Good news here. I reinstalled Ubuntu Server 20.04.2 64bits on RPi4B, then just install all the softwares needed and compile the workspace **without upgrading any [...]

Hi @mchobbylong

I have reviewed all process and it is working. It was a great opportunity to review all procedure, and there are some changes, to improve the process. ej. ros-desktop is not more necessary.

I was reviewing our conversation and I realised that the command to add the user to the group is wrong.

Try with it,.
```
sudo usermod -a -G i2c ubuntu
```

You can use the ' getent group i2c' to validate it.
```
i2c:x:120:ubuntu
```

mike4192, 2021-05-18:
@jpereirajr 

Here is a basic diagram around the PCA9685 of how I power it. Servos are ultimately powered via a direct connection to a 2s lipo battery, while the circuit components operate at a 5v level from a regulated 5v power supply that also powers other devices like the Raspberry Pi. Need to make sure everything ultimately has a common ground.

I shorted the power inputs on the PCA9685 board as shown to the servo power rails. This may not be necessary at all, but at the time I read other people talk about a diode or fuse being inbetween the voltage input and servo rails. This may or may not be true. 

[image]

kennydedavion, 2021-06-13:
[image]
Good day. 
When I try to run the spot I still encounter the i2cpwm_board problem.
I have tried all the procedures listed here and still without the possibility of communication. 
Disassembly of Uno, ultrasonic sensor, and oled display, due to the possibility of influencing errors. 
Mpu6050 and bmp280 connected to pca9685, connected (SDA, SCL, GND, VCC), without possibility of disconnection. 
Level converter added between rpi3 for further elimination of operating errors. 
Adapter 5V 3500mAh. Access error. Please help

zoltan0907, 2022-01-22:
@mchobbylong

Not sure if it is currently interesting to you but regarding:

```
[ INFO] [1619295376.431822812]: I2C bus opened on /dev/i2c-1
[ INFO] [1619295376.446875080]: Setting PWM frequency to 50 Hz
[FATAL] [1619295377.457911782]: Failed to open I2C bus (null)
[ INFO] [1619295377.460355694]: Setting PWM frequency to 50 Hz
[1]    11746 segmentation fault (core dumped)  rosrun i2cpwm_board i2cpwm_board
```

I checked out the latest commit of https://gitlab.com/bradanlane/ros-i2cpwmboard which includes the changes for smbus and i2c and i got the same behaviour. But when compiling the single package i got a warning that _load_params didn't return a value, but the function-def says it should return an `int`. 

```
Warnings   << i2cpwm_board:make /home/ubuntu/catkin_ws/logs/i2cpwm_board/build.make.007.log                                                                                    
/home/ubuntu/catkin_ws/src/ros-i2cpwmboard/src/i2cpwm_controller.cpp: In function ‘int _load_params()’:
/home/ubuntu/catkin_ws/src/ros-i2cpwmboard/src/i2cpwm_controller.cpp:1571:1: warning: no return statement in function returning non-void [-Wreturn-type]
 1571 | }
      | ^
```

I added a simple `return 0;` at the end of the function and compile the package again and the statrtup succeeds.

sffranke, 2022-03-05:
Hi all,

I tried the fork too. All looks fine, calibraton reacts normal. But controlling the robot with the PS4 Controller it reacts seconds delayed and the servos have little torque only.

16BitSingh, 2022-06-20:
Hi all, just reopening this discussion. It seems the link to the ros-i2cpwmboard is no longer active and produces a 404 when followed. https://gitlab.com/bradanlane/ros-i2cpwmboard.
@[jacklu333333](https://github.com/jacklu333333) recommended using [this submodule link](https://github.com/mentor-dyun/ros-i2cpwmboard) but I havent quite tested if this works yet.

PhuongV05, 2024-03-23:
Hi, 
I am trying to run "roslaunch spot_micro_keyboard_command keyboard_command.launch run_plot:=true", the keyboard command works normally, but matplotlib displays a blank animation, the window is just white. Upon killing the spot_micro_plot_node, the window displays the plot of the bot but it cannot receive any inputs from the keyboard command. However, I can still move the camera about and interact with the plot.
How can I get the plot to display properly? I am running on a Raspberry Pi 4 and connecting to it from a Ubuntu VM via SSH.