mike4192/spotMicro #61: Spot Micro is not building in catkin workspace

b-madhukar, 2021-03-20:


b-madhukar, 2021-03-20:
ubuntu@ubiquityrobot:~/catkin_ws/spotMicro$ catkin build spot_micro_motion_cmd
-----------------------------------------------------------
Profile:                     default
Extending:             [env] /opt/ros/kinetic
Workspace:                   /home/ubuntu/catkin_ws
-----------------------------------------------------------
Build Space:        [exists] /home/ubuntu/catkin_ws/build
Devel Space:        [exists] /home/ubuntu/catkin_ws/devel
Install Space:      [unused] /home/ubuntu/catkin_ws/install
Log Space:         [missing] /home/ubuntu/catkin_ws/logs
Source Space:       [exists] /home/ubuntu/catkin_ws/src
DESTDIR:            [unused] None
-----------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
-----------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-----------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-----------------------------------------------------------
Workspace configuration appears valid.
-----------------------------------------------------------
[build] Found '4' packages in 0.0 seconds.
[build] Given package 'spot_micro_motion_cmd' is not in the workspace

b-madhukar, 2021-03-20:
build] Given package 'spot_micro_motion_cmd' is not in the workspace

mike4192, 2021-03-20:
The contents of this repo should be in your `/home/ubuntu/catkin_ws/src` directory. 

It looks like you have it instead in `/home/ubuntu/catkin_ws/spotMicro`

Can you please provide more information. Simply pasting your log output is not very helpful. What are you building on (rpi or a pc?) Which OS of linux? What ROS version do you have installed.

b-madhukar, 2021-03-23:
Board : Raspberry pi3 model B+
OS: Downloaded from Ubiquity robotics version : 2020-11-07-ubiquity-xenial-lxde

Ros came with catkin workspace when i try to build its showing command not found, can i get any complete documentation
[image]
[image]

mike4192, 2021-03-24:
Did you move the repo to a `src` directory within `catkin_ws`?

Sorry I don't have any additional documentation. I suggest trying to follow the ROS tutorials and see if you can get those working on your raspberry pi: http://wiki.ros.org/ROS/Tutorials

That may give you better insight how to get this project to build.