mike4192/spotMicro #63: how to start for a newbie

liwenz, 2021-03-26:
I hope to start the project.
I have installed ros  kenetic on ubuntu16 in vmware and learned tutorial for ros. I could not  find ubuntu 16 for pi 3B and install pi os and compile ros kenetic on it(ok).
Now I try to compile on ubuntu 16 at first.
in ~/catkin_ws/src

leon@ubuntu:~/catkin_ws/src$ git init
Initialized empty Git repository in /home/leon/catkin_ws/src/.git/
leon@ubuntu:~/catkin_ws/src$ git pull https://github.com/mike4192/spotMicro

now the file and directory is as follow:
`leon@ubuntu:~/catkin_ws/src$ ls ros-i2c*

leon@ubuntu:~/catkin_ws/src$ ls
assets          cmake.txt    learning_launch   learning_topic  ros-i2cpwmboard      spot_micro_keyboard_command  spot_micro_plot
CHANGELOG.md    docs         learning_service  LICENSE         servo_move_keyboard  spot_micro_launch            spot_micro_rviz
CMakeLists.txt  lcd_monitor  learning_tf       README.md       spot_micro_joy       spot_micro_motion_cmd
leon@ubuntu:~/catkin_ws/src$`   

leon@ubuntu:~/catkin_ws$ git submodule update --init --recursive
fatal: Not a git repository (or any of the parent directories): .git
leon@ubuntu:~/catkin_ws$ sudo apt-get install libi2c-dev
[sudo] password for leon: 
ok

leon@ubuntu:~/catkin_ws$ catkin_make
error:
-- Could not find the required component 'i2cpwm_board'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):

I tried:
https://github.com/dheera/ros-pwm-pca9685
sudo apt-get install libi2c-dev done

how to resolve no 'i2cpwm_board'?

liwenz, 2021-03-26:
wget from url = https://gitlab.com/bradanlane/ros-i2cpwmboard.git
then :
leon@ubuntu:~/catkin_ws/src$ unzip -l r*.zip
Archive:  ros-i2cpwmboard-master.zip
9d5b00da8dbdbc542343b65f87f505b0ddb33f8f

leon@ubuntu:~/catkin_ws/src/ros-i2cpwmboard-master$ ls
CMakeLists.txt  doc  launch  msg  package.xml  README.md  src  srv

but:
leon@ubuntu:~/catkin_ws$ catkin_make --only-pkg-with-deps ros-i2cpwmboard-master
Base path: /home/leon/catkin_ws
Source space: /home/leon/catkin_ws/src
Build space: /home/leon/catkin_ws/build
Devel space: /home/leon/catkin_ws/devel
Install space: /home/leon/catkin_ws/install
Packages "ros-i2cpwmboard-master" not found in the workspace

liwenz, 2021-03-26:
sudo apt-get install ros-kinetic-joy
sudo apt-get install ros-kinetic-rplidar-ros
sudo apt-get install ros-kinetic-hector-slam

then joy is ok
but 

-- +++ processing catkin package: 'spot_micro_joy'
-- ==> add_subdirectory(spot_micro_joy)
-- +++ processing catkin package: 'spot_micro_motion_cmd'
-- ==> add_subdirectory(spot_micro_motion_cmd)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
CMake Error at spot_micro_motion_cmd/CMakeLists.txt:20 (add_subdirectory):
  The source directory

    /home/leon/dog_ws/src/spot_micro_motion_cmd/libs/spot_micro_kinematics_cpp

  does not contain a CMakeLists.txt file.

liwenz, 2021-03-26:
git submodule command should be in src folder:
At first, it said some is not free. I mv it,then it's ok as follow:
 
leon@ubuntu:~/dog_ws/src$ git submodule update --init --recursive
Cloning into 'ros-i2cpwmboard'...
remote: Enumerating objects: 10, done.
remote: Counting objects: 100% (10/10), done.
remote: Compressing objects: 100% (9/9), done.
remote: Total 258 (delta 2), reused 0 (delta 0), pack-reused 248
Receiving objects: 100% (258/258), 261.32 KiB | 0 bytes/s, done.
Resolving deltas: 100% (126/126), done.
Checking connectivity... done.
Submodule path 'ros-i2cpwmboard': checked out 'c73f89a8687e2a9c7698522d6015e06f1228050e'
Cloning into 'spot_micro_motion_cmd/libs/spot_micro_kinematics_cpp'...
remote: Enumerating objects: 207, done.
remote: Counting objects: 100% (207/207), done.
remote: Compressing objects: 100% (126/126), done.
remote: Total 207 (delta 91), reused 162 (delta 51), pack-reused 0
Receiving objects: 100% (207/207), 40.61 KiB | 0 bytes/s, done.
Resolving deltas: 100% (91/91), done.
Checking connectivity... done.
Submodule path 'spot_micro_motion_cmd/libs/spot_micro_kinematics_cpp': checked out 'ef154d7c05eba241800319cbdf58d0d437dbb072'
Cloning into 'spot_micro_plot/scripts/spot_micro_kinematics_python'...
remote: Enumerating objects: 104, done.
remote: Counting objects: 100% (104/104), done.
remote: Compressing objects: 100% (54/54), done.
remote: Total 104 (delta 55), reused 90 (delta 47), pack-reused 0
Receiving objects: 100% (104/104), 694.39 KiB | 0 bytes/s, done.
Resolving deltas: 100% (55/55), done.
Checking connectivity... done.
Submodule path 'spot_micro_plot/scripts/spot_micro_kinematics_python': checked out '29efae700b73c3d970d1ada4a81d0f93d53ff680'
leon@ubuntu:~/dog_ws/src$ cd ..

then catkin_make is ok:

leon@ubuntu:~/dog_ws$ catkin_make
Base path: /home/leon/dog_ws
Source space: /home/leon/dog_ws/src
Build space: /home/leon/dog_ws/build
Devel space: /home/leon/dog_ws/devel
Install space: /home/leon/dog_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/leon/dog_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/leon/dog_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/leon/catkin_ws/devel;/opt/ros/kinetic
-- This workspace overlays: /home/leon/catkin_ws/devel;/opt/ros/kinetic
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.12", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python2
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/leon/dog_ws/build/test_results
-- Found gtest sources under '/usr/src/gmock': gtests will be built
-- Found gmock sources under '/usr/src/gmock': gmock will be built
-- Found PythonInterp: /usr/bin/python2 (found version "2.7.12") 
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.29
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 9 packages in topological order:
-- ~~  - i2cpwm_board
-- ~~  - lcd_monitor
-- ~~  - servo_move_keyboard
-- ~~  - spot_micro_keyboard_command
-- ~~  - spot_micro_plot
-- ~~  - spot_micro_rviz
-- ~~  - spot_micro_joy
[... 124 lines cut ...]
[ 81%] Built target i2cpwm_board_generate_messages
Scanning dependencies of target test_spot_micro_kinematics
[ 82%] Building CXX object spot_micro_motion_cmd/libs/spot_micro_kinematics_cpp/CMakeFiles/test_spot_micro_kinematics.dir/tests/test_spot_micro_kinematics.cpp.o
[ 83%] Linking CXX executable test_spot_micro_kinematics
[ 85%] Building CXX object spot_micro_motion_cmd/CMakeFiles/smfsm.dir/src/smfsm/spot_micro_stand.cpp.o
[ 85%] Built target test_spot_micro_kinematics
Scanning dependencies of target test_spot_micro_leg
[ 86%] Building CXX object spot_micro_motion_cmd/libs/spot_micro_kinematics_cpp/CMakeFiles/test_spot_micro_leg.dir/tests/test_spot_micro_leg.cpp.o
[ 87%] Linking CXX executable test_spot_micro_leg
[ 87%] Built target test_spot_micro_leg
Scanning dependencies of target test_utils
[ 88%] Building CXX object spot_micro_motion_cmd/libs/spot_micro_kinematics_cpp/CMakeFiles/test_utils.dir/tests/test_utils.cpp.o
[ 90%] Building CXX object spot_micro_motion_cmd/CMakeFiles/smfsm.dir/src/smfsm/spot_micro_transition_stand.cpp.o
[ 91%] Linking CXX executable test_utils
[ 91%] Built target test_utils
[ 92%] Building CXX object spot_micro_motion_cmd/CMakeFiles/smfsm.dir/src/smfsm/spot_micro_transition_idle.cpp.o
[ 93%] Building CXX object spot_micro_motion_cmd/CMakeFiles/smfsm.dir/src/smfsm/spot_micro_walk.cpp.o
[ 95%] Linking CXX shared library /home/leon/dog_ws/devel/lib/libsmfsm.so
[ 95%] Built target smfsm
Scanning dependencies of target spot_micro_motion_cmd_node
[ 96%] Building CXX object spot_micro_motion_cmd/CMakeFiles/spot_micro_motion_cmd_node.dir/src/spot_micro_motion_cmd.cpp.o
[ 97%] Building CXX object spot_micro_motion_cmd/CMakeFiles/spot_micro_motion_cmd_node.dir/src/spot_micro_motion_cmd_node.cpp.o
[ 98%] Building CXX object spot_micro_motion_cmd/CMakeFiles/spot_micro_motion_cmd_node.dir/src/utils.cpp.o
[100%] Linking CXX executable /home/leon/dog_ws/devel/lib/spot_micro_motion_cmd/spot_micro_motion_cmd_node
[100%] Built target spot_micro_motion_cmd_node

liwenz, 2021-03-26:
leon@ubuntu:~$ **mkdir -p dog/src**
leon@ubuntu:~$ **cd dog**
leon@ubuntu:~/dog$ ls
src
leon@ubuntu:~/dog$ **cd src**
leon@ubuntu:~/dog/src$ **git init**
Initialized empty Git repository in /home/leon/dog/src/.git/
leon@ubuntu:~/dog/src$  **git pull https://github.com/mike4192/spotMicro**
remote: Enumerating objects: 32, done.
remote: Counting objects: 100% (32/32), done.
remote: Compressing objects: 100% (26/26), done.
remote: Total 1152 (delta 11), reused 15 (delta 5), pack-reused 1120
Receiving objects: 100% (1152/1152), 25.59 MiB | 3.45 MiB/s, done.
Resolving deltas: 100% (608/608), done.
From https://github.com/mike4192/spotMicro
 * branch            HEAD       -> FETCH_HEAD
leon@ubuntu:~/dog/src$ git **submodule update --init --recursive**
Submodule 'ros-i2cpwmboard' (https://gitlab.com/bradanlane/ros-i2cpwmboard.git) registered for path 'ros-i2cpwmboard'
Submodule 'spot_micro_motion_cmd/libs/spot_micro_kinematics_cpp' (https://github.com/mike4192/spot_micro_kinematics_cpp.git) registered for path 'spot_micro_motion_cmd/libs/spot_micro_kinematics_cpp'
Submodule 'spot_micro_plot/scripts/spot_micro_kinematics_python' (https://github.com/mike4192/spot_micro_kinematics_python.git) registered for path 'spot_micro_plot/scripts/spot_micro_kinematics_python'
Cloning into 'ros-i2cpwmboard'...
remote: Enumerating objects: 10, done.
remote: Counting objects: 100% (10/10), done.
remote: Compressing objects: 100% (9/9), done.
remote: Total 258 (delta 2), reused 0 (delta 0), pack-reused 248
Receiving objects: 100% (258/258), 261.32 KiB | 0 bytes/s, done.
Resolving deltas: 100% (126/126), done.
Checking connectivity... done.
Submodule path 'ros-i2cpwmboard': checked out 'c73f89a8687e2a9c7698522d6015e06f1228050e'
Cloning into 'spot_micro_motion_cmd/libs/spot_micro_kinematics_cpp'...
remote: Enumerating objects: 207, done.
remote: Counting objects: 100% (207/207), done.
remote: Compressing objects: 100% (126/126), done.
remote: Total 207 (delta 91), reused 162 (delta 51), pack-reused 0
Receiving objects: 100% (207/207), 40.61 KiB | 0 bytes/s, done.
Resolving deltas: 100% (91/91), done.
Checking connectivity... done.
Submodule path 'spot_micro_motion_cmd/libs/spot_micro_kinematics_cpp': checked out 'ef154d7c05eba241800319cbdf58d0d437dbb072'
Cloning into 'spot_micro_plot/scripts/spot_micro_kinematics_python'...
remote: Enumerating objects: 104, done.
remote: Counting objects: 100% (104/104), done.
remote: Compressing objects: 100% (54/54), done.
remote: Total 104 (delta 55), reused 90 (delta 47), pack-reused 0
Receiving objects: 100% (104/104), 694.39 KiB | 0 bytes/s, done.
Resolving deltas: 100% (55/55), done.
Checking connectivity... done.
Submodule path 'spot_micro_plot/scripts/spot_micro_kinematics_python': checked out '29efae700b73c3d970d1ada4a81d0f93d53ff680'
leon@ubuntu:~/dog/src$ **git submodule update --recursive**
leon@ubuntu:~/dog/src$ **sudo apt-get install ros-kinetic-joy**

liwenz, 2021-03-26:
I tried to compile on raspberry  os PI b3,
ros--geometry-msgs needed.
and  
sudo apt-get install ros-kinetic-joy
sudo apt-get install ros-kinetic-rplidar-ros
sudo apt-get install ros-kinetic-hector-slam
can't be found
I only compiled the core version ros and no vision.

mike4192, 2021-03-27:
I wouldn't recommend trying to compile ros from source on raspbian OS, I tried it early on and ran into issues. It also takes very long to compile. The issues can be overcome but it will probably be tedious and a long drawn out process. If you wanted to continue this way, you would also need to compile those packages from source as well in your catkin ws src directory (joy, rplidar, hector-slam, etc...).

Why could you not utilize premade ubuntu RPi images from ubinquit robotics like I recommend in the documentation? These come preinstalled with ros basically ready to go, and then you can install the required packages via `sudo apt-get` install no problemo.

https://downloads.ubiquityrobotics.com/pi.html

liwenz, 2021-03-29:
I have ompiled it  100% on ubuntu18 (download from: https://releases.ubuntu-mate.org/archived/18.04/arm64/)
ros/melodic ( I install it as http://wiki.ros.org/melodic/Installation/Ubuntu)

it is almost same as I did last post.
except:
sudo apt-get install ros-melodic-joy
sudo apt-get install ros-melodic-rplidar-ros
sudo apt-get install ros-melodic-hector-slam

the i2ccpwm need change as git:
https://github.com/EricWiener/ros-i2cpwmboards/commit/9ae9556a043ac2fe75a3251c9b28318b5383e744

Next I need test and let it walk.

mike4192, 2021-04-02:
Hmm, I sometimes see this error  but I'm not sure what causes it. Usually if I see it I try relaunching, or power cycling. I would verify you are powering all the boards with the correct voltages and have a common ground for everything.

You can also try running the python script for moving servos with the i2cpwmboard library. 

`roslaunch servo_move_keyboard keyboard_move.launch`

mike4192, 2021-04-02:
You can run the code somewhat like a simulation by launching the following:

On the raspberry pi:
`roslaunch spot_micro_motion_cmd motion_cmd.launch run_standalone:=true`

On a computer:
`roslaunch spot_micro_keyboard_command keyboard_command.launch run_rviz:=true`

You should then be able to control the robot software and see it walk around in the visualization.

liwenz, 2021-04-02:
Thanks， I'll try tomorrow. I'm in Canada. It's 23:15

liwenz, 2021-04-02:
I will change this issue as how to start (compile) and open a new issue as how to start (test and run it)

closed 2021-06-25 by mike4192