mike4192/spotMicro #64: how to test and run it for a newbie?

liwenz, 2021-04-03:
Make sure you can see the pwm board on your RPi. See the following page, section "Testing I2C":
https://learn.adafruit.com/adafruits-raspberry-pi-lesson-4-gpio-setup/configuring-i2c#testing-i2c-2724591-14
For example, running the command like: sudo i2cdetect -y 1

leon@leon-desktop:~$ sudo i2cdetect -y 1
[sudo] password for leon:
Error: Could not open file `/dev/i2c-1' or `/dev/i2c/1': No such file or directory
leon@leon-desktop:~$

Also, you are running the i2cpwmboard on the raspberry pi right? And not on your PC?

pi b3

roslaunch servo_move_keyboard keyboard_move.launch

...
setting /run_id to c3180844-941a-11eb-b0be-b827eb8897a3
process[rosout-1]: started with pid [3482]
started core service [/rosout]
process[servo_move_keyboard_node-2]: started with pid [3485]
[INFO] [1617412834.089777]: > Servos corrrectly initialized
[INFO] [1617412834.107231]: > Publisher corrrectly initialized
[INFO] [1617412834.116593]: Initialization complete

Servo Control Module for 12 Servos.

Enter one of the following options:
-----------------------------
quit: stop and quit the program
oneServo: Move one servo manually, all others will be commanded to their center position
allServos: Move all servo's manually together

Keyboard commands for One Servo Control
---------------------------
   q                y
            f   g       j   k
    z   x       b   n   m

  q: Quit current command mode and go back to Option Select
  z: Command servo min value
  y: Command servo center value
  x: Command servo max value
  f: Manually decrease servo command value by 10
  g: Manually decrease servo command value by 1
  j: Manually increase servo command value by 1
  k: Manually increase servo command value by 10
  b: Save new min command value
  n: Save new center command value
  m: Save new max command value

  anything else : Prompt again for command

CTRL-C to quit

Command?:

I don't know how to choose the server number or load the parameter.
Is there a link?

liwenz, 2021-04-03:
3 command total
**No. 1:
roslaunch spot_micro_motion_cmd motion_cmd.launch run_standalone:=true**

started roslaunch server http://leon-desktop:46693/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.10
 * /spot_micro_motion_cmd_node/LB_1/center: 306
 * /spot_micro_motion_cmd_node/LB_1/center_angle_deg: -0.4
 * /spot_micro_motion_cmd_node/LB_1/direction: -1
 * /spot_micro_motion_cmd_node/LB_1/num: 9
 * /spot_micro_motion_cmd_node/LB_1/range: 374
 * /spot_micro_motion_cmd_node/LB_2/center: 306
 * /spot_micro_motion_cmd_node/LB_2/center_angle_deg: 38.7
 * /spot_micro_motion_cmd_node/LB_2/direction: 1
 * /spot_micro_motion_cmd_node/LB_2/num: 8
 * /spot_micro_motion_cmd_node/LB_2/range: 392
 * /spot_micro_motion_cmd_node/LB_3/center: 306
 * /spot_micro_motion_cmd_node/LB_3/center_angle_deg: -73.9
 * /spot_micro_motion_cmd_node/LB_3/direction: 1
 * /spot_micro_motion_cmd_node/LB_3/num: 7
 * /spot_micro_motion_cmd_node/LB_3/range: 390
 * /spot_micro_motion_cmd_node/LF_1/center: 306
 * /spot_micro_motion_cmd_node/LF_1/center_angle_deg: -7.6
 * /spot_micro_motion_cmd_node/LF_1/direction: 1
 * /spot_micro_motion_cmd_node/LF_1/num: 12
 * /spot_micro_motion_cmd_node/LF_1/range: 389
 * /spot_micro_motion_cmd_node/LF_2/center: 306
 * /spot_micro_motion_cmd_node/LF_2/center_angle_deg: 38.6
 * /spot_micro_motion_cmd_node/LF_2/direction: 1
 * /spot_micro_motion_cmd_node/LF_2/num: 11
 * /spot_micro_motion_cmd_node/LF_2/range: 397
 * /spot_micro_motion_cmd_node/LF_3/center: 306
 * /spot_micro_motion_cmd_node/LF_3/center_angle_deg: -82.8
 * /spot_micro_motion_cmd_node/LF_3/direction: 1
 * /spot_micro_motion_cmd_node/LF_3/num: 10
 * /spot_micro_motion_cmd_node/LF_3/range: 387
 * /spot_micro_motion_cmd_node/RB_1/center: 306
 * /spot_micro_motion_cmd_node/RB_1/center_angle_deg: -4.4
 * /spot_micro_motion_cmd_node/RB_1/direction: 1
 * /spot_micro_motion_cmd_node/RB_1/num: 6
 * /spot_micro_motion_cmd_node/RB_1/range: 411
 * /spot_micro_motion_cmd_node/RB_2/center: 306
 * /spot_micro_motion_cmd_node/RB_2/center_angle_deg: -35.4
 * /spot_micro_motion_cmd_node/RB_2/direction: 1
 * /spot_micro_motion_cmd_node/RB_2/num: 5
 * /spot_micro_motion_cmd_node/RB_2/range: 396
 * /spot_micro_motion_cmd_node/RB_3/center: 306
 * /spot_micro_motion_cmd_node/RB_3/center_angle_deg: 85.8
 * /spot_micro_motion_cmd_node/RB_3/direction: 1
 * /spot_micro_motion_cmd_node/RB_3/num: 4
 * /spot_micro_motion_cmd_node/RB_3/range: 396
 * /spot_micro_motion_cmd_node/RF_1/center: 306
 * /spot_micro_motion_cmd_node/RF_1/center_angle_deg: -5.4
 * /spot_micro_motion_cmd_node/RF_1/direction: -1
 * /spot_micro_motion_cmd_node/RF_1/num: 3
 * /spot_micro_motion_cmd_node/RF_1/range: 396
 * /spot_micro_motion_cmd_node/RF_2/center: 306
 * /spot_micro_motion_cmd_node/RF_2/center_angle_deg: -27.6
 * /spot_micro_motion_cmd_node/RF_2/direction: 1
 * /spot_micro_motion_cmd_node/RF_2/num: 2
 * /spot_micro_motion_cmd_node/RF_2/range: 389
 * /spot_micro_motion_cmd_node/RF_3/center: 306
 * /spot_micro_motion_cmd_node/RF_3/center_angle_deg: 88.2
 * /spot_micro_motion_cmd_node/RF_3/direction: 1
[... 61 lines cut ...]

**No. 3
roslaunch i2cpwm_board i2cpwm_node.launch**
on pi 3B 
from : http://bradanlane.gitlab.io/ros-i2cpwmboard/

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.10

NODES
  /
    i2cpwm_board_node (i2cpwm_board/i2cpwm_board)

auto-starting new master
process[master]: started with pid [3584]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to fffe69ea-941c-11eb-b0be-b827eb8897a3
process[rosout-1]: started with pid [3595]
started core service [/rosout]
process[i2cpwm_board_node-2]: started with pid [3598]
sudo: no tty present and no askpass program specified
[i2cpwm_board_node-2] process has died [pid 3598, exit code 1, cmd sudo /home/leon/dog/devel/lib/i2cpwm_board/i2cpwm_board __name:=i2cpwm_board_node __log:=/home/leon/.ros/log/fffe69ea-941c-11eb-b0be-b827eb8897a3/i2cpwm_board_node-2.log].
log file: /home/leon/.ros/log/fffe69ea-941c-11eb-b0be-b827eb8897a3/i2cpwm_board_node-2*.log

mike4192, 2021-04-03:
If spot_micro_motion_cmd in standalone mode, you do not need to run the i2cpwm_board node.

I think you need to verify you can exchange communication between your ubuntu virtual machine and the raspberry pi.

They must be connected on the same network, or to eachother. You may need to modify network settings in the virtual machine software to get communication with the raspberry pi.

First verify you can ping the ip address of the raspberry pi from your ubuntu virtual machine. Find what ip address the raspbery pi has on it's wireless network interface through `ifconfig`.

On the virtual machine try pinging that ip address, example: `ping 192.168.1.xx` where xx is the last part of the address the pi has.

On your virtual machine you need to make sure you set some environment variables so nodes know the ROS master is on the pi. You may need to set ROS_MASTER_URI and ROS_IP. I have the following in my ~/.bashrc file. You'd have to replace http://ubunturospi.local with the ip address of your raspberry pi. You would also need to set ROS_IP to your virtual machine's ip address.

`export ROS_MASTER_URI=http://ubunturospi.local:11311 # Set IP of ros master, raspberry pi in this case
export ROS_IP=$(hostname -I | cut -f1 -d' ') # set own ros ip environment variable

alias rosinit="source /home/mike/catkin_ws/devel/setup.bash && export ROS_WORKSPACE=~/catkin_ws"
`

Before trying to run this software, I would make sure you can successfully run a ROS tutorial example, where one node is on the Rapsberry pi, and one node is on the virtual machine. For example, the simple publisher and subscriber: http://wiki.ros.org/ROS/Tutorials/WritingPublisherSubscriber%28python%29

The following video may help guide you as a basic ros tutorial for running stuff on a raspberr pu: https://www.youtube.com/watch?v=iLiI_IRedhI

liwenz, 2021-04-04:
Thanks for your reply!
My pc(win10) and pi are the same wifi network and I use ssh to control pi on win10. But my vm (on win10) has some problem.
Maybe I played the docker then vm ubuntu16 lost local network. At first,VM ubuntu 16 could not work then I close Hyper-V and vm rework.
I don't know how to fix it.

The robot should work on only one pi . Maybe more terminal. Then remote node control mode.
I bought 8g pi 4b on March 13, but  Amazon delay to next week.
At first, i2c servo should work. The servo rotate on my zynq 7010 system(Xilinx fpga) by direct pwm and i2c two way.

Or it should be simulate on a ROS and I understand Kinematics.
https://spotmicroai.readthedocs.io/en/latest/simulation/ seems a good site.

AlekseyErmolaev, 2021-04-06:
Hey! I didn't create a new topic. I hope you can help me.
I have a small problem and I cannot figure out where to look for a solution.
I have RPi4(ubiquityrobot), ubuntu 16.04 laptop, terminal with ssh. catkin_make is 100% assembled.
I use `roslaunch spot_micro_keyboard_command keyboard_command.launch run_rviz:=true` on laptop and `roslaunch spot_micro_motion_cmd motion_cmd.launch run_standalone:=true` on RPi.
After the command "stand" appears:

[ WARN] [1617730100.302570705]: Joint LF_2, Angle: 37.42
[ WARN] [1617730100.302659575]: Proportional Command below -1.0 was computed, clipped to -1.0
[ WARN] [1617730100.302758926]: Joint LF_3, Angle: -128.00
[ WARN] [1617730100.302853740]: Proportional Command above +1.0 was computed, clipped to 1.0
[ WARN] [1617730100.302923776]: Joint RB_3, Angle: 128.14
[ WARN] [1617730100.303019090]: Proportional Command above +1.0 was computed, clipped to 1.0
[ WARN] [1617730100.303093126]: Joint RF_3, Angle: 128.00
[ WARN] [1617730100.322427290]: Proportional Command below -1.0 was computed, clipped to -1.0
[ WARN] [1617730100.322514511]: Joint LB_3, Angle: -128.15
[ WARN] [1617730100.322585288]: Proportional Command above +1.0 was computed, clipped to 1.0

Angle values change. Joint numbers do not change.
In rviz, the image of the legs begins to twitch.
I really hope for your help!

mike4192, 2021-04-07:
Did you modify the configuration at all? This warning means an angle beyond the intended travel range was computed for some reason.

If the legs are twitching they may inadvertently be another node trying to command the leg servos.

AlekseyErmolaev, 2021-04-08:
Here is a screenshot of my table
[image]
Maybe you can see what's wrong

liwenz, 2021-04-10:
I reinstall ubuntu mate 16(pi) and vm ubuntu 16 (another pc)
pi ip: 192.168.2.80
vm ip:192.168.2.85
but I don't know the remote publisher, 
What should start in pi, and what should do on vm?

roslaunch servo_move_keyboard keyboard_move.launch
it seems to work well.
No error, and I know choose the servo number, and x,y,z show the angles.

Should the servo run to min,center,max, or only a config?
My servo did not run

liwenz, 2021-04-11:
roslaunch servo_move_keyboard keyboard_move.launch
it seems run ok as follow, but servo did not move, then logic analyzer check it. There is no wave.
From doc, it should control the servo move and there should be waves.
 
Which servo to control? Enter a number 1 through 12: 1
Enter command, q to go back to option select:
Servo  0 cmd:  520
Servo  0 cmd:  510
Servo  0 cmd:  500
Servo  0 cmd:  490
Servo  0 cmd:  480
Servo  0 cmd:  470
Servo  0 cmd:  460
Servo  0 cmd:  450
Servo  0 cmd:  440
Servo  0 cmd:  430
Servo  0 cmd:  420
Servo  0 cmd:  410
Servo  0 cmd:  400
Servo  0 cmd:  306
Servo  0 cmd:   83
Servo  0 cmd:  520
Servo  0 cmd:  520
Servo  0 cmd:  510
Servo  0 cmd:  500
Servo  0 cmd:  490
Servo  0 cmd:  489
Servo  0 cmd:  488
Servo  0 cmd:  487
Servo  0 cmd:  486
Key not in valid key commands, try again
Servo  0 cmd:  476
Servo  0 cmd:  466
Servo  0 cmd:  465
Servo  0 cmd:  464
Servo  0 cmd:  465
Servo  0 cmd:  466
Servo  0 cmd:  476
Servo  0 cmd:  486
Servo  0 cmd:  496

mike4192, 2021-04-13:
@liwenz you should first verify your VM can communicate with the raspberry pi. For example, try pinging the pi from your VM:
`ping 192.168.2.80`
and vice versa from the raspberry pi. If not, you need to figure that out first.

Next, try running an example publisher and subscriber in ROS, where one node is running on the raspberry pi and the other on the VM, and see if you can get that to work: http://wiki.ros.org/ROS/Tutorials/WritingPublisherSubscriber%28python%29

mike4192, 2021-04-13:
@AlekseyErmolaev your config values look reasonable, or at least they don't have any obvious errors, but they are very different from my set, which is the default set on this repo. However the values are entirely dependent on the positions a servo is installed.

I would double check you followed the calibration directions. Try spot checking a few servos using your calibration table. For example, for the right front lower leg joint (the knee), according to your config table if you command it to a pwm value of 306, will the lower leg link be position truly at 0 degrees?

liwenz, 2021-04-13:
ping and netccat  from pi to vm are ok now.

Remote machine topic listed, but no data transmitted

mike4192, 2021-04-13:
Which topic?

liwenz, 2021-04-13:
I forgot, maybe turtle_cmd, I check it again.
publisher, the turtle move.

liwenz, 2021-04-13:
my test come from: http://wiki.ros.org/ROS/Tutorials/MultipleMachines

ssh hal
roscore

ssh marvin
export ROS_MASTER_URI=http://hal:11311
rosrun rospy_tutorials listener.py
py => rosrun turtlesim turtlesim_node

ssh hal
export ROS_MASTER_URI=http://hal:11311
rosrun rospy_tutorials talker.py
py=> rostopic pub -1 /turtle1/cmd_vel geometry_msgs/Twist -- '[2.0, 0.0, 0.0]' '[0.0, 0.0, 1.8]'

it could control the turtle move  in one pc or pi
but remote could not work

rpstopic list work well.

leon@leon:~$ rostopic list
/rosout
/rosout_agg
/turtle1/cmd_vel
/turtle1/color_sensor
/turtle1/pose
leon@leon:~$

liwenz, 2021-04-22:
roslaunch servo_move_keyboard keyboard_move.launch 
the launch file includes:
<node if="$(arg run_i2c_pwmboard)" name="i2cpwm_board_node" pkg="i2cpwm_board" type="i2cpwm_board" output="screen">  $
    </node>
how to set "$(arg run_i2c_pwmboard)" ?
I think I did not start it, so i2cpwm_board did not start.

mike4192, 2021-04-24:
Not sure what to tell you, running your command works for me, specifically:
`roslaunch servo_move_keyboard keyboard_move.launch run_i2c_pwmboard:=true`

Note, however, the i2c_pwmboard node will only work correctly if run on a raspberry pi with the pwm board installed and with the i2c, power, and ground pins correctly connected.

Is your pca9685 board powered correctly?. The board requires power both for the board circuitry, and for the servo power. See the following diagram for example, note the top power input from an external 5v power source.
https://steemitimages.com/1280x0/https://res.cloudinary.com/hpiynhbhq/image/upload/v1518024645/eh21urwjil2ply3ay2gz.png

The pi alone probably will not supply enough power to allow the servos to move.

liwenz, 2021-04-29:
roslaunch servo_move_keyboard keyboard_move.launch run_i2c_pwmboard:=true
it works well. Now I could control  servo.
What's next? led display?
or  roslaunch spot_micro_motion_cmd motion_cmd.launch run_standalone:=false

kennydedavion, 2021-06-04:
Hi, I also have a problem, more.
I am attaching a picture. According to the possibility from the IT specialist at Ros, I had to install a version of melodic.
according to the instruction on the instructable "Rpi4 Lidar melodic Ros".
I can't run anything right now.

i2cpwm_board display reports error:
[FATAL] [1622811777.479435609]: Failed to open I2C bus (null)

spot_micro_motion_command reports an error:
RLException: [spot_micro_motion_cmdConfig-version.cmakerostopic] is neither a launch file in package [spot_micro_motion_cmd] nor is [spot_micro_motion_cmd] a launch file name
The traceback for the exception was written to the log file.

Although everything is in place, but the different versions without catkin have changed, and no one is able to tell me now how to run it?
ps: all the repair options I found and tried, either didn't help or decided the system so that I had to format the storage and reinstall everything.

Thanks for your instructions. There is no arduino in your spot? do you control the rpi servo-shield?
All the problems that, Arduino UNO, Rpi3b and Raspbian with ROS Melodic, contain with this mode, and there is no person who knows all three of these topics, so there is no help for me.

I'm trying to find a solution to combine Ros in such a way that it reads and writes everything to UNO just like the Arduino IDE and its SerialMonitor.

Any ideas to help?
Jacob

[image]

liwenz, 2021-08-17:
where could I find the geometry data, as leg lenth, the arm coordinate，angle.