mike4192/spotMicro #65: Add sensors to the model

JonathanMortes, 2021-06-09:
Hi, I have added some extra sensors to my spot mini and I'd like to know how to add them to the model so it han handle and be trained with those. 
Thanks

mike4192, 2021-06-11:
Hi. This is a very vague and open ended question that has no direct answer, so I'm afraid I can't help you much, but I'll share what I'd do.

Question what purpose you want to achieve by adding a sensor, which will determine what kinds of sensor you need, how many, and where and how it should be mounted. A sensor will have a unique interface that needs to be handled. It may communicate via i2c, serial, spi, usb, or maybe even ethernet. I'd look for any existing ROS libraries or packages that handle similar sensors (are the sensor drivers) for guidance on how to use them.

If by training you mean using machine learning to do something, that's really vague and open ended, you'd need to come up with a specific goal to do something. For example, adding an IMU to attempt to utilize reinforcement learning to develop a walking gait would require setting up a simulation and whole learning/training pipeline. Adding a camera to do image classification is probably a lot easier because you could probably find a ton of existing libraries.

JonathanMortes, 2021-06-11:
Hi Mike, thanks for the answers!.

 I’ve added some force sensors and IMUs to the legs to get better information about its movement, and as you said, my initial idea was to include them on the model for RL. I believe it only uses the one Central body IMU and maybe 4 force/hall sensors on the foot (from what I could see on the spot_real codes)but please correct me if i’m wrong.

Knowing this, id share my case and doubt of usage in case you might be able to help me:

I’m finishing  my masters degree project with spotmicroai and my idea is to communicate via ros and do Control + extract all sensor info and after reading the paper and seeing the project, I was very interested in using RL for the gait.

I’m using an ESP32 as an interface for low level control inside the robot and I’m able to comunicate via serial and/or wifi with a linux system (RPi / PC) for the heavy lifting. My goal is to use the spot_real with RL as a starting point and further down the line develope the model and extras. But I really did not get how the software runs on the RPi on your implementation. I understood that The board on spot reads and handles sensors via serial as well as handling the bidirectional communication with the motors. But I’d love to know/understand better how does the RPi handle this info and apply the trained model with RL for the gait control (I believe it’s part of the launch file) but if you could explain it a bit it would be wonderful and really appreciated. This way I could iterate on it and develope a new approach for communication that I will share and cite on the publication.

And sure using artificial vision would be easier down the line and I’ve got plans on developing on mine.

thanks in advance for your feedback, help and time. It’s really appreciated ☺️

closed 2021-06-12 by JonathanMortes