mike4192/spotMicro #66: Robot keeps falling when walking .. MG996R servos

pringithub, 2021-06-25:
Hi, I've built two of these robots, one with and one without the shoulders. We're using TowerPro MG996R servos and, while the robots can both stand fine, they can not get up without assistance, and once they start walking they will also fall.

I can't tell if the poor performance is due to the servo torque not being high enough or due to a calibration issue .. The PCA9685 is receiving 5.3V but the voltage also varies throughout for every pin due to inherent drops across the board .. the 470uF cap might night be that strong either.

According to the datasheet, the MG996R has a stall torque around 10 kg-cm at 5.3V. 
The PDI-HV5523MG, on the other hand, has a stall torque around 18 kg-cm at 5.3V.
This may explain why my servos are stalling out while other builders that use the latter servos are doing fine. However, the original KDY short-body seems to use MG996R, though there is a later revision for CLS6336HV which have a much higher stall torque. https://spotmicroai.readthedocs.io/en/latest/originalReadme/

So .. do I have to buy these expensive servos???

https://www.electronicoscaldas.com/datasheet/MG996R_Tower-Pro.pdf
http://www.jx-servo.com/en/Product/STANDARD/SD/550.html

mike4192, 2021-06-25:
Hi.

Yes it's because of the servos. I also originally built a first iteration with MG966R servos, and they were not powerful enough to operate the robot. I think mine could barely stand, but if it tried walking it was wobbly and all over the place, basically unusable.

I would not trust stall torque specifications of cheap servos. All I can say is I an multiple other people have built the robot with the PDI servos and achieved acceptable performance.

pringithub, 2021-06-25:
Dang .. $$$$

Well thanks for the quick response!!

pringithub, 2021-06-25:
I noticed you also wired your battery directly to the servo board. The servo board has a max input voltage of 5.5V, and I'm assuming your battery was 7.4V .. did you have any problems with this?

mike4192, 2021-06-26:
See the following comment where I posted a mini power wiring diagram.
https://github.com/mike4192/spotMicro/issues/59#issuecomment-843516142

The PDI servos support a direct 7.4v input. The servo board sort of has two power inputs. The top power block I believe is only connected to the servo power rails and a common ground. The side input power pin is the circuit power input, and probably has a max of 5v or so, this input is not meant to power the servos.

closed 2021-08-14 by mike4192