mike4192/spotMicro #68: Could not find a package configuration file provided by "hector_geotiff"   with any of the following names:      hector_geotiffConfig.cmake     hector_geotiff-config.cmake

b-madhukar, 2021-07-27:
Base path: /home/ubuntu/catkin_ws
Source space: /home/ubuntu/catkin_ws/src
Build space: /home/ubuntu/catkin_ws/build
Devel space: /home/ubuntu/catkin_ws/devel
Install space: /home/ubuntu/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/ubuntu/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/ubuntu/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/ubuntu/catkin_ws/devel;/opt/ros/kinetic
-- This workspace overlays: /home/ubuntu/catkin_ws/devel;/opt/ros/kinetic
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.12", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python2
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/ubuntu/catkin_ws/build/test_results
-- Found gtest sources under '/usr/src/gmock': gtests will be built
-- Found gmock sources under '/usr/src/gmock': gmock will be built
-- Found PythonInterp: /usr/bin/python2 (found version "2.7.12") 
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.29
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
WARNING: package "dnn_detect" should not depend on metapackage "image_transport_plugins" but on its packages instead
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 14 packages in topological order:
-- ~~  - dnn_rotate
-- ~~  - docking
-- ~~  - fiducial_follow
-- ~~  - magni_lidar
-- ~~  - move_demo
-- ~~  - i2cpwm_board
-- ~~  - lcd_monitor
-- ~~  - servo_move_keyboard
-- ~~  - spot_micro_keyboard_command
-- ~~  - spot_micro_plot
-- ~~  - spot_micro_rviz
-- ~~  - spot_micro_joy
-- ~~  - spot_micro_motion_cmd
-- ~~  - spot_micro_launch
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'dnn_rotate'
-- ==> add_subdirectory(demos/dnn_rotate)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- dnn_rotate: 0 messages, 1 services
-- +++ processing catkin package: 'docking'
-- ==> add_subdirectory(demos/docking)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- docking: 0 messages, 1 services
-- +++ processing catkin package: 'fiducial_follow'
-- ==> add_subdirectory(demos/fiducial_follow)
-- +++ processing catkin package: 'magni_lidar'
-- ==> add_subdirectory(demos/magni_lidar)
-- +++ processing catkin package: 'move_demo'
-- ==> add_subdirectory(demos/move_demo)
-- +++ processing catkin package: 'i2cpwm_board'
-- ==> add_subdirectory(spotMicro/ros-i2cpwmboard)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- i2cpwm_board: 6 messages, 4 services
-- +++ processing catkin package: 'lcd_monitor'
-- ==> add_subdirectory(spotMicro/lcd_monitor)
-- +++ processing catkin package: 'servo_move_keyboard'
-- ==> add_subdirectory(spotMicro/servo_move_keyboard)
-- +++ processing catkin package: 'spot_micro_keyboard_command'
-- ==> add_subdirectory(spotMicro/spot_micro_keyboard_command)
-- +++ processing catkin package: 'spot_micro_plot'
-- ==> add_subdirectory(spotMicro/spot_micro_plot)
-- +++ processing catkin package: 'spot_micro_rviz'
-- ==> add_subdirectory(spotMicro/spot_micro_rviz)
-- +++ processing catkin package: 'spot_micro_joy'
-- ==> add_subdirectory(spotMicro/spot_micro_joy)
-- +++ processing catkin package: 'spot_micro_motion_cmd'
-- ==> add_subdirectory(spotMicro/spot_micro_motion_cmd)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- +++ processing catkin package: 'spot_micro_launch'
-- ==> add_subdirectory(spotMicro/spot_micro_launch)
CMake Warning at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package):
  Could not find a package configuration file provided by "hector_geotiff"
  with any of the following names:

    hector_geotiffConfig.cmake
    hector_geotiff-config.cmake

  Add the installation prefix of "hector_geotiff" to CMAKE_PREFIX_PATH or set
  "hector_geotiff_DIR" to a directory containing one of the above files.  If
  "hector_geotiff" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  spotMicro/spot_micro_launch/CMakeLists.txt:10 (find_package)

-- Could not find the required component 'hector_geotiff'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "hector_geotiff"
  with any of the following names:

    hector_geotiffConfig.cmake
    hector_geotiff-config.cmake

  Add the installation prefix of "hector_geotiff" to CMAKE_PREFIX_PATH or set
  "hector_geotiff_DIR" to a directory containing one of the above files.  If
  "hector_geotiff" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  spotMicro/spot_micro_launch/CMakeLists.txt:10 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/ubuntu/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/ubuntu/catkin_ws/build/CMakeFiles/CMakeError.log".
Makefile:808: recipe for target 'cmake_check_build_system' failed
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed

mike4192, 2021-08-14:
Have you installed the required dependent ros packages? See readme for:
```
sudo apt-get install ros-kinetic-joy
sudo apt-get install ros-kinetic-rplidar-ros
sudo apt-get install ros-kinetic-hector-slam
```

Hlebario, 2023-08-07:
how to do it with windows 10?