mike4192/spotMicro #71: How do I use the OpenSource Code in this Github Project?

RedgarOpenSource, 2021-08-23:
Hello Spot Micro AI developers! I am so in love with this project and really want to build and try it for myself. Before I build it, I really wanted to have a concrete set of steps laid out for myself. While I did read the documentation which included assembly instructions and saw the video of the robot, I still don't understand the process behind uploading the different kinds of code on this page. I couldn't find assembly instructions including the code. I wanted to use the Raspberry Pi 4 and Arduino for this project. Are there any tutorials I may not have seen or any instructions on what kind of code goes to what device? Thank you so much in advance! I am a beginner at robotics as I am more of a software person!

MZandtheRaspberryPi, 2021-08-23:
Hello. I would say a simpler project may be a better jumping point into robots like a kit that comes with all parts you need and more fleshed out instructions.

That said, if you really want to build this one, Mike4192 wrote some comments in the main README.md that links to a page with instructions on uploading an image to the raspberry pi 4: https://github.com/mike4192/spotMicro. Its under the header "software". I used a PI4 and had some troubles with the wifi, so added some bits to the instructions to reflect that.

I did a blog post about it here that may be useful, but its not a step by step guide: https://github.com/MZandtheRaspberryPi/spot_micro_demo

And then there's the main spot micro site which does have step by step for assembly but not software. https://spotmicroai.readthedocs.io/en/latest/

mike4192, 2021-08-24:
I Echo @MZandtheRaspberryPi comments.

Projects like this rarely have comprehensive step by step procedures, however a similar project that does have more information is the stanford quadruped pupper, which I think is also built on a raspberry pi. It, however, is reliant on custom fabricated parts and more expensive servos, and is not built on ROS, but it's documentation may be useful to understanding this project. https://pupper.readthedocs.io/en/latest/

In general, you should learn how to use linux, how to use a linux machine through SSH, learn about ROS, and try running example ROS code on a raspberry pi before attempting to build anything.

RedgarOpenSource, 2021-08-24:
@MZandtheRaspberryPi and @mike4192 Thank you so much for your responses! I will look more into ROS and another simpler Quadruped Robotics project. I have a fairly decent experience with Linux and Raspberry Pi Development but clearly not enough to understand how to work with all of the code in this repository. I also would like to maybe start with the Stanford Pupper before this kind of project is worked on. I also do have tons of experience with 3D Printing and 3D Modeling so the Stanford Pupper should be better suited for me as a beginner. I would like to say that this project is still really awesome and I really want to follow along until I understand these things better. Thank you so much for your feedback!