mike4192/spotMicro #72: How can I simulate the motion of robot?

robo4869, 2021-08-28:
Hi @mike4192, spotMicro developers!

I'm quite new here. I'm so excited about this project and really want to build and try it for myself. Now, I'm making a code for controlling the motion of my fake robot via the keyboard. Because I wanted to have a concrete set of steps, so I read your documentation, but I don't know how to run that type of control with your robot, like the gif below. How can I do?

Thanks.
[image]

robo4869, 2021-08-28:
I mean, I just want to control the fake robot, not by displays a 3d model of the robot with state updates in real-time. Unfortunately, because of COVID-19, I can't make a real robot, so everything is simulated.

robo4869, 2021-08-29:
Can I have your email address? I'll contact you via email because my question is not really an issue.

robo4869, 2021-09-01:
@sbias it seems mike didn't see this issue, could you help me? Or can I have your email address?

mike4192, 2021-09-05:
Unfortunately I'm not really sure what you are asking. I think you are asking how to control a simulated version of the robot, but without this repos control software? This is not exactly trivial, and unfortunately I am not willing to hand hold someone through the process of setting it up as it would take a substantial amount of time.

I will leave you with at least the following info:

- First off, there is no true "simulation" of the robot in this repo, where by simulation I mean a physics model of the robot, a ground interaction model, servos/actuators model, sensors model, etc... If you want to control a true physics based simulation of the spot micro platform walking on ground, you will need to look elsewhere as this repo doesn't provide it.
- The visualization this repo supports, which I may refer to as a "simulation" in some of the documentation, is purely a visualization of joint states and transform states (e.g. ROS tf states). Both joint states, and purely simulated odometry which propogates the robots position, are outputted by the main motion control node
- If you want to control the visualization (e.g. drive it with your own joint states and transform states) you will need to read up on ROS TF and joint states for background info, and then create your own node that publishes similar tf messages as the motion control module does.
- If you just want to create an alternative keyboard control node, then you would just create a different version of the spot_micro_keyboard_command node that yields the same functionality.

robo4869, 2021-09-05:
I saw you say: " `run_rviz:=true` Starts RVIZ and displays a 3d model of the robot with state update in real-time." But this time I don't have a real one, then I ask to do a fake robot that can do some gaits by following a designed trajectory.

I found another repo and can do rest_mode over a keyboard (they use joystick). 

Thanks for replying.

closed 2021-09-05 by robo4869