mike4192/spotMicro #73: Failed to process package 'spot_micro_motion_cmd':

kennydedavion, 2021-09-03:
I do the construction exactly according to the instructions and I keep showing me this at catkin make:
 [ 96%] Building CXX object CMakeFiles/spot_micro_motion_cmd_node.dir/src/utils.cpp.o
/home/pi/ros_catkin_ws/src/spot_micro_motion_cmd/src/utils.cpp:6:10: fatal error: f2_eigen/tf2_eigen.h: Adresář nebo soubor neexistuje
 #include "tf2_eigen/tf2_eigen.h"
          ^~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/spot_micro_motion_cmd_node.dir/build.make:104: CMakeFiles/spot_micro_motion_cmd_node.dir/src/utils.cpp.o] Chyba 1
make[2]: *** Čeká se na nedokončené úlohy…
make[1]: *** [CMakeFiles/Makefile2:205: CMakeFiles/spot_micro_motion_cmd_node.dir/all] Chyba 2
make: *** [Makefile:146: all] Chyba 2
<== Failed to process package 'spot_micro_motion_cmd': 
  Command '['/opt/ros/melodic/env.sh', 'make', '-j2']' returned non-zero exit status 2

Reproduce this error by running:
==> cd /home/pi/ros_catkin_ws/build_isolated/spot_micro_motion_cmd && /opt/ros/melodic/env.sh make -j2

Command failed, exiting.

Of course, the above remedy does not do anything unexpected and throws out the same error. dear Mike, even if the library tf and tf2 in src i have how to fix this so that the system finds dependence? even rosdep is not able to handle this addiction properly.

mike4192, 2021-09-05:
It appears you need to install eigen c++ dependency on your computer. If you are on ubuntu or debian based linux you could potentially do this via `sudo apt install libeigen3-dev`. If you are on a different linux distro you will need to find other ways to install.

You will then need to make sure your ros environment path includes the location of eigen header files on your system. When eigen is installed via command as above, header files are typically installed in typical locations and should be accessible via the default system path. But you may have additional issues beyond my control.

kennydedavion, 2021-09-05:
Yes, it will probably be out of your control, because this package is already installed on raspbian : libeigen3-dev je již nejnovější verze (3.3.7-1). 
this is the result of catkin build:
- Starting  >>> tf2                                                                                                                                
Finished  <<< visualization_msgs                              [ 5.4 seconds ]                                                                    
Finished  <<< roscpp                                          [ 5.3 seconds ]                                                                    
Starting  >>> rospy                                                                                                                              
Starting  >>> bondcpp                                                                                                                            
Finished  <<< rospy                                           [ 1.2 seconds ]                                                                    
Finished  <<< tf2                                             [ 3.0 seconds ]                                                                    
Finished  <<< bondcpp                                         [ 3.4 seconds ]                                                                    
Starting  >>> rosserial_python                                                                                                                   
Starting  >>> rqt_gui                                                                                                                            
Starting  >>> tf2_bullet                                                                                                                         
Finished  <<< control_msgs                                    [ 9.1 seconds ]                                                                    
Starting  >>> tf2_eigen                                                                                                                          
Finished  <<< tf2_bullet                                      [ 0.9 seconds ]                                                                    
Starting  >>> urdfdom_py                                                                                                                         
Finished  <<< rqt_gui                                         [ 0.9 seconds ]                                                                    
Starting  >>> bondpy                                                                                                                             
Finished  <<< rosserial_python                                [ 0.8 seconds ]                                                                    
Starting  >>> rosserial_xbee                                                                                                                     
Finished  <<< tf2_eigen                                       [ 0.8 seconds ]                                                                    

I'll probably deal with ROS, because for an ordinary mortal who wants to create and not constantly make some mistakes, this is not the case. In two years, I understood the Arduino, the peripherals, the communication with them and the overall control, (although I had to pre-program the steps of the servants for the poor) in Arduino goes and reads and writes data over the serial line. Now, since January, I have been working on ROS and so far only three tutorials that have no flaws, bugs or problems: cloning, building, and running. I2c pwm board not running, due to access issues, your SpotMicro is also reporting new bugs, spotMini, and similarly do not use rpi3 or arduino and many of the tutorials and procedures I need are outdated and inoperative. I'm most embarrassed by the instructions, where it's easy to install with a display: sudo apt install rosdistro- anything, and the package can't be found. Thanks to the answare, I have repeatedly convinced myself that IT experts on this midleware actually have no idea how this works, and after a series of similar advisors, I had over 350 packages in the workspace, and none of them could compile into a build. Thanks for your work and effort, but your stuck in kinetics compared to the current version is unforgettable. It is a pity that this midleware still has so many problems that there is no solution, because vctools and wcstools must be used, and it still does not complete the installation of dependencies and packages, because it cannot clean new things and always rebuild everything because the display needs to compile everything, instead of checks and possible modifications, as opposed to deletion. Fixes that turn out to be an option don't work, as if the installer didn't know he had done the display now. Missing dependencies, although the package is already compiled, as I add in the list above. Well, if someone reports it and there will be no problems with these recurring problems even with a new clean image installation, I'll be happy to throw myself into it, but I refuse to contribute to this mess. Thx.

closed 2021-09-12 by mike4192

kennydedavion, 2022-09-21:
the problem is still not solved, due to the incompetence of the builder and referring elsewhere