mike4192/spotMicro #74: Changing robot size

sokol07, 2021-09-18:
Hi,
I would like to build a spotMicro, however, my 3d printer is slightly too small to print the largest parts.
How much work would it be to change the spotMicro software to work with a smaller body (the general mechanics are the same, the legs configuration and DOFs are the same, just the lengths/heights/widths will change)? The electronics would remain the same (of course, different servos, but the only difference would be, again, its size)
Generally, I know what is going on in the software as I studied robotics so kinematics and reverse kinematics aren't unknown for me, however, I haven't worked with ROS before and I am not sure if I understand correctly what is going on in which part of the project.
Is it possible to point out which parts of the project (which files?) need to be modified to change the robot dimensions?
Do I guess correctly that the whole kinematics magic, which would be affected by the size change, is happening in the spot_micro_kinematics modules?
I guess that also the urdf model of the robot for rviz would change (probably I should also change the slt to my models).
Something else?
Thanks in advance - I am really impressed by your work, the code is very well documented and clear even to someone totally new to the topic. Not mentioning that the things, which this codes does, are very cool... ;)

mike4192, 2021-09-19:
Hi. It should definitely be possible to utilize this repo with modified robot dimensions. As long as the differences aren't crazy large, it should be perfectly fine.

Before going further I would note that the mechanics of the robot are a big time sink (nuts and bolts and servo sizes and installation, wiring, etc.. as well as CAD file modifications that would be needed to resize) so if you can avoid modifications I'd recommend it.

Regardless, in general, robot dimensions are only specified in a few places:

- [The config file for spot_micro_motion_command node](https://github.com/mike4192/spotMicro/blob/master/spot_micro_motion_cmd/config/spot_micro_motion_cmd.yaml) holds robot dimensions which can be modified (e.g.: lower_leg_link_length)
- The [URDF file](https://github.com/mike4192/spotMicro/blob/master/spot_micro_rviz/urdf/spot_micro.urdf.xacro) would also need modification. This one also has robot dimensions variables near the top, but I forget what all represent. Note, the STL file visualization would probably be screwed up, but if you don't care about accuracy it'll probably still visually make enough sense to understand what the robot is doing.
- If using spot_micro_plot node for plotting a stick figure, the hard coded robot dimension values in [this file](https://github.com/mike4192/spot_micro_kinematics_python/blob/29efae700b73c3d970d1ada4a81d0f93d53ff680/spot_micro_stick_figure.py) would need to be changed.

sokol07, 2021-09-20:
Thanks for reply!
The differences aren't extreme, the most important thing is that the kinematics are the same, only the dimensions differ. I think it would be a nice project anyway to adapt MicroSpotAI code to work with smaller mechanics for poor people with tiny printers. ;)
I have a complete robot design, adapted for smaller servos (9g tiny ones). However, its author created his robot using ESP32, which I have never liked and I decided to use his mechanics but MicroSpot's software. I know that it will take some work to adapt it but I believe this can be done as the differences aren't that big.

I have one more question about the URDF file - is it used only for visualization or is it used for calculations too? The needs of changes in the first file in obvious, however, what about the URDF file - is it necessary to change it for proper work or is it needed only for visualization and can be left as is if the visualization accuracy isn't crucial?