mike4192/spotMicro #77: Raspberry Pi images preloaded with Ubuntu 16.04 and a ROS Kinetic installation failed

Janeyw, 2022-02-02:
Hi,
I downloaded Raspberry Pi images preloaded with Ubuntu 16.04 and a ROS Kinetic installation to SD card to setup RPI but failed. 
Can anyone help to solve this problem? I used "2021-11-24-ubiquity-xenial-lxde" version.

thank you

chrisalbertson, 2022-02-02:
Maybe you can describe the steps you took and the nature of the failure.

I have Ubuntu 20 on my robot control Pi4.  I took a different route.  I got
"Ubuntu Server" from the Ubuntu web site.  Made some user accounts, got it
running on my network, and then added the rest of the software.

Janeyw, 2022-02-02:
> Maybe you can describe the steps you took and the nature of the failure. I have Ubuntu 20 on my robot control Pi4. I took a different route. I got "Ubuntu Server" from the Ubuntu web site. Made some user accounts, got it running on my network, and then added the rest of the software.
> […](#)
> -- Chris Albertson Redondo Beach, California

Hi Chris,
Thanks for your reply! The author mentioned the software might not work on versions other than Ubuntu 16.04 and the Ros framework. You didn't encounter any issue with Ubuntu 20, right? The problem I encountered is when tried to setup my RPI3 b+ it shows 
unable to mount root fs on unknown block (179,2)

chrisalbertson, 2022-02-02:
The images on the official Ubuntu website generally work.  Many people have
used them.

The usual problem with self-made images is that the person who made them
changes something about his Pi and forget about it or it could be the SD
card is bad or many other things.