mike4192/spotMicro #88: ESP32 version :)

antonioasaro, 2023-01-01:
I couldn't get my hands on a Pi4 so I decided to try to port everything to the ESP32.
It took quite a while to get everything to work. 

https://github.com/antonioasaro/spotmicro_ws

chrisalbertson, 2023-01-01:
How is your "Spot" walking?  Is it smooth and accurate yet?  I seem to be
having a ton of trouble with the PCA9685 dropping data when pushed hard.
 Also, even when working fine, the pulse width resolution is poor.   I'm
building a replacement servo controller that runs on Raspberry Pico.   It
will accept joint angles in radians, in the robot's joint frame and then
apply per servo calibration, enforce limits and then set the PWM.     All
in Micropython.   The code is to yet finished.     Micropython is very
impressive.  They even have a numpy port

Also, I notice many "Spots" have a mistake in inverse kinematics.  Their
code assumes (wrongly) that the 1st and second axis intersect.  They
don't.   On SpotMicro the second axis is about 16 mm below the first.  so
the leg has in effect FOUR, not three links and angles with one angle fixed
at 90 degrees.   You can ignore this but then you never get accurate foot
placement.   The same IK code keeps getting re-used and this error spreads.

antonioasaro, 2023-01-01:
Hi Chris,

    Here's a link to a YouTube video showing SpotMicro walking.
    I think it's fairly smooth.
    I did have issues with movement  - e.g. staggered, jumpy.
    But, I believe these issues were more power delivery related - e.g. servos drawing too many amps???

    Anyways, I didn't write any of "base" SpotMicro code.
    I only modified mike1492's code where I had too - e.g. #ifdef out unsupported ROS functions.
    But I did review it and AFAICT, the inverse kinematics equations were similar to what I've seen others do.

    Initially, my SpotMicro was taking some steps and falling over, like a drunk.
    But after a full calibration, it was much better - again, see YouTube video link.
    The calibration data goes into a spreadsheet that creates the params.h for the servos.

Hope that helps,
--Tony

https://www.youtube.com/watch?v=lUxBdqUD2ac
[https://www.bing.com/th?id=OVF.LPpykRwvCadQrGXLOkiiWw&pid=Api]<https://www.youtube.com/watch?v=lUxBdqUD2ac>
SpotMicro Walking<https://www.youtube.com/watch?v=lUxBdqUD2ac>
SpotMicro Walking (after following the proper calibration steps)
www.youtube.com