mike4192/spotMicro #90: how to additional ROS packages having windows 10

Hlebario, 2023-08-07:
I have the problems. I have Windows 10 , I need additional ros-kinetic-joy, ros-kinetic-rplidar-ros, ros-kinetic-hector-slam to the project spotMicro. how to do it?

chrisalbertson, 2023-08-08:
I’m working on a SpotMicro derivative too.   Realistically just do not bother with Windows 10.  With Linux, the software is mature and well tested and thousands of people can answer questions.  

OK, if you MUST use Windows 10 then you need to be more specific.  Tell us what happened when you tried to run the keyboard joystick node.  What went wrong, what have you tried already?  Perhaps you could cut and paste some error messages you got.  And do start with res-kinetic-joy.  Get just this to work first and I susspect the same solution will work for th others.  But “joy” is so simple, get the simplest thins to work first.

BTW.  How is the spot micro working for you?   I am finding servos to be too slow and started working on a BLDC motor design.  Which variant are you building?

But really, unless you have a VERY good reason to use Windows, don’t.

Hlebario, 2023-08-08:
Thank you for taking the time to respond. I'm new to this case and my mistakes will seem simple to you. I'm building a robot for myself. I just started and wanted to figure out how it works on a computer, and then order components, so I wanted to start building a project in VS Code. I tried it in the Ubuntu emulator for Windows 10 (WSL) - it did not work, when I entered the command `sudo apt update `(paragraph 1.4 of the official instructions), I received a 404 server error response. I switched to Windows 10, there were a lot of errors during assembly, namely: the absence of packages `ros-kinetic-rplidar-ros, ros-kinetic-joy, ros-kinetic-hector-slam`, the requirement of `std:c++20` (although indicated in `c_cpp_properties.json` os++20). I'll attach a photos. As for the BLDC motors, I like the idea because the servos are noisy. But the motors are more expensive and they need speed regulators. Maybe a hydraulic system?))
[image]
[image]
[image]
[image]

[image]

chrisalbertson, 2023-08-09:
You are just lain going to have to abandon the use of Windows for
robotics.   This means also any kind of emulator.

It is also odd that you are using ROS for a Spot Micro.  Most people are
using stand-alone code on a Raspberry Pi or a microcontroller.

But if you are using RS Kinetic, it will be simple under Linux.