mike4192/spotMicro #94: i2c_smbus_read_byte_data’ was not declared in this scope; did you mean ‘i2c_smbus_ioctl_data

LEIJUNMIUI, 2023-11-01:
error: ‘i2c_smbus_read_byte_data’ was not declared in this scope; did you mean ‘i2c_smbus_ioctl_data’?
  422 |     oldmode = i2c_smbus_read_byte_data (_controller_io_handle, __MODE1);
      |               ^~~~~~~~~~~~~~~~~~~~~~~~
      |               i2c_smbus_ioctl_data
/home/sunrise/catkin_ws/src/ros-i2cpwmboard/src/i2cpwm_controller.cpp:425:13: error: ‘i2c_smbus_write_byte_data’ was not declared in this scope; did you mean ‘i2c_smbus_ioctl_data’?
  425 |     if (0 > i2c_smbus_write_byte_data (_controller_io_handle, __MODE1, newmode)) // go to sleep
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~
      |             i2c_smbus_ioctl_data
/home/sunrise/catkin_ws/src/ros-i2cpwmboard/src/i2cpwm_controller.cpp:428:14: error: ‘i2c_smbus_write_byte_data’ was not declared in this scope; did you mean ‘i2c_smbus_ioctl_data’?
  428 |     if (0 >  i2c_smbus_write_byte_data(_controller_io_handle, __PRESCALE, (int)(floor(prescale))))
      |              ^~~~~~~~~~~~~~~~~~~~~~~~~
      |              i2c_smbus_ioctl_data
/home/sunrise/catkin_ws/src/ros-i2cpwmboard/src/i2cpwm_controller.cpp:431:13: error: ‘i2c_smbus_write_byte_data’ was not declared in this scope; did you mean ‘i2c_smbus_ioctl_data’?
  431 |     if (0 > i2c_smbus_write_byte_data(_controller_io_handle, __MODE1, oldmode))
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~
      |             i2c_smbus_ioctl_data
/home/sunrise/catkin_ws/src/ros-i2cpwmboard/src/i2cpwm_controller.cpp:436:13: error: ‘i2c_smbus_write_byte_data’ was not declared in this scope; did you mean ‘i2c_smbus_ioctl_data’?
  436 |     if (0 > i2c_smbus_write_byte_data(_controller_io_handle, __MODE1, oldmode | 0x80))
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~
      |             i2c_smbus_ioctl_data
/home/sunrise/catkin_ws/src/ros-i2cpwmboard/src/i2cpwm_controller.cpp: In function ‘void _set_pwm_interval_all(int, int)’:
/home/sunrise/catkin_ws/src/ros-i2cpwmboard/src/i2cpwm_controller.cpp:459:13: error: ‘i2c_smbus_write_byte_data’ was not declared in this scope; did you mean ‘i2c_smbus_ioctl_data’?
  459 |     if (0 > i2c_smbus_write_byte_data (_controller_io_handle, __ALL_CHANNELS_ON_L, start & 0xFF))
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~
      |             i2c_smbus_ioctl_data
/home/sunrise/catkin_ws/src/ros-i2cpwmboard/src/i2cpwm_controller.cpp:461:14: error: ‘i2c_smbus_write_byte_data’ was not declared in this scope; did you mean ‘i2c_smbus_ioctl_data’?
  461 |     if (0 >  i2c_smbus_write_byte_data (_controller_io_handle, __ALL_CHANNELS_ON_H, start  >> 8))
      |              ^~~~~~~~~~~~~~~~~~~~~~~~~
      |              i2c_smbus_ioctl_data
/home/sunrise/catkin_ws/src/ros-i2cpwmboard/src/i2cpwm_controller.cpp:463:13: error: ‘i2c_smbus_write_byte_data’ was not declared in this scope; did you mean ‘i2c_smbus_ioctl_data’?
  463 |     if (0 > i2c_smbus_write_byte_data (_controller_io_handle, __ALL_CHANNELS_OFF_L, end & 0xFF))
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~
      |             i2c_smbus_ioctl_data
/home/sunrise/catkin_ws/src/ros-i2cpwmboard/src/i2cpwm_controller.cpp:465:13: error: ‘i2c_smbus_write_byte_data’ was not declared in this scope; did you mean ‘i2c_smbus_ioctl_data’?
  465 |     if (0 > i2c_smbus_write_byte_data (_controller_io_handle, __ALL_CHANNELS_OFF_H, end >> 8))
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~
      |             i2c_smbus_ioctl_data
/home/sunrise/catkin_ws/src/ros-i2cpwmboard/src/i2cpwm_controller.cpp: In function ‘void _set_active_board(int)’:
/home/sunrise/catkin_ws/src/ros-i2cpwmboard/src/i2cpwm_controller.cpp:501:21: error: ‘i2c_smbus_write_byte_data’ was not declared in this scope; did you mean ‘i2c_smbus_ioctl_data’?
  501 |             if (0 > i2c_smbus_write_byte_data (_controller_io_handle, __MODE2, __OUTDRV))
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~~
      |                     i2c_smbus_ioctl_data
/home/sunrise/catkin_ws/src/ros-i2cpwmboard/src/i2cpwm_controller.cpp:504:21: error: ‘i2c_smbus_write_byte_data’ was not declared in this scope; did you mean ‘i2c_smbus_ioctl_data’?
  504 |             if (0 > i2c_smbus_write_byte_data (_controller_io_handle, __MODE1, __ALLCALL))
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~~
      |                     i2c_smbus_ioctl_data
/home/sunrise/catkin_ws/src/ros-i2cpwmboard/src/i2cpwm_controller.cpp:510:24: error: ‘i2c_smbus_read_byte_data’ was not declared in this scope; did you mean ‘i2c_smbus_ioctl_data’?
  510 |             mode1res = i2c_smbus_read_byte_data (_controller_io_handle, __MODE1);
      |                        ^~~~~~~~~~~~~~~~~~~~~~~~
      |                        i2c_smbus_ioctl_data
/home/sunrise/catkin_ws/src/ros-i2cpwmboard/src/i2cpwm_controller.cpp:513:21: error: ‘i2c_smbus_write_byte_data’ was not declared in this scope; did you mean ‘i2c_smbus_ioctl_data’?
  513 |             if (0 > i2c_smbus_write_byte_data (_controller_io_handle, __MODE1, mode1res))
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~~
      |                     i2c_smbus_ioctl_data
[ 78%] Built target spot_micro_kinematics
/home/sunrise/catkin_ws/src/ros-i2cpwmboard/src/i2cpwm_controller.cpp: In function ‘void _set_pwm_interval(int, int, int)’:
/home/sunrise/catkin_ws/src/ros-i2cpwmboard/src/i2cpwm_controller.cpp:555:13: error: ‘i2c_smbus_write_byte_data’ was not declared in this scope; did you mean ‘i2c_smbus_ioctl_data’?
  555 |     if (0 > i2c_smbus_write_byte_data (_controller_io_handle, __CHANNEL_ON_L+4*channel, start & 0xFF))
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~
      |             i2c_smbus_ioctl_data
Scanning dependencies of target test_spot_micro_leg
/home/sunrise/catkin_ws/src/ros-i2cpwmboard/src/i2cpwm_controller.cpp:557:14: error: ‘i2c_smbus_write_byte_data’ was not declared in this scope; did you mean ‘i2c_smbus_ioctl_data’?
  557 |     if (0 >  i2c_smbus_write_byte_data (_controller_io_handle, __CHANNEL_ON_H+4*channel, start  >> 8))
      |              ^~~~~~~~~~~~~~~~~~~~~~~~~
      |              i2c_smbus_ioctl_data
Scanning dependencies of target smfsm
Scanning dependencies of target test_spot_micro_kinematics
/home/sunrise/catkin_ws/src/ros-i2cpwmboard/src/i2cpwm_controller.cpp:559:13: error: ‘i2c_smbus_write_byte_data’ was not declared in this scope; did you mean ‘i2c_smbus_ioctl_data’?
  559 |     if (0 > i2c_smbus_write_byte_data (_controller_io_handle, __CHANNEL_OFF_L+4*channel, end & 0xFF))
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~
      |             i2c_smbus_ioctl_data
[ 79%] Building CXX object spot_micro_motion_cmd/libs/spot_micro_kinematics_cpp/CMakeFiles/test_spot_micro_leg.dir/tests/test_spot_micro_leg.cpp.o
/home/sunrise/catkin_ws/src/ros-i2cpwmboard/src/i2cpwm_controller.cpp:561:13: error: ‘i2c_smbus_write_byte_data’ was not declared in this scope; did you mean ‘i2c_smbus_ioctl_data’?
  561 |     if (0 > i2c_smbus_write_byte_data (_controller_io_handle, __CHANNEL_OFF_H+4*channel, end >> 8))

when I compile the ros project in horizon x3 pi, this bug occured!

HLkyss, 2024-02-24:
same issue

HLkyss, 2024-02-24:
just solved by adding some changes from this work: https://github.com/jpereirajr/spotMicro